Ysurac/openmptcprouter #2436: No connections After update wrt3200acs

domoclick, 2022-07-24:
## Expected Behavior

## Expected Behavior

I Just upgrade from v.0.58.5.r0, full working, to v.0.59.rc1.5.4 in my linksys wrt3200acm but after upgrade no internet connections and i cant ping anything, even 1.1.1.1 or 8.8.8.8. Wan seems to work because i see traffico in and out but i cant see any sites.

## Current Behavior

If i install a fresh versione 0.58, configure my wans, internet Is ok and everything work. I flash new firmware 0.59, configure my wans and nothing works, only lan side Is ok.
Can you help me please?

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country: 

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2022-07-24:
Would be better with a completed template...
What you have in status page ?

domoclick, 2022-07-24:
hi, Ysurac and thanks for answer.

this is my working Dashboard
[image]

Then, i flash new version openmptcprouter-v0.59rc1-5.4-r0+16594-ce92de8c8c-mvebu-cortexa9-linksys_wrt3200acm-squashfs-sysupgrade, no keep setting, reconfigure my wan1 with static ip, i see my public ip, but i cant ping anything and no internet connection up.
[image]

> Would be better with a completed template... What you have in status page ?

Ysurac, 2022-07-24:
I would need to see the status page. In System->OpenMPTCProuter, "status" tab.

domoclick, 2022-07-24:
Ysurac, i just rebooted my router and now it works!!

closed 2022-07-24 by domoclick