Ysurac/openmptcprouter #2437: Building OMR for new tag v0.59rc1 unsupported platform failed (X86_64)

iamdivyanshverma, 2022-07-25:
## Expected Behavior

## Current Behavior

divyansh@divyansh-Latitude-7480:~/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter$ OMR_TARGET="x86_64" OMR_FEED_SRC="v0.59rc1" ./build.sh -j4
remote: Enumerating objects: 4284, done.
remote: Counting objects: 100% (2126/2126), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 4284 (delta 2112), reused 2122 (delta 2112), pack-reused 2158
Receiving objects: 100% (4284/4284), 1.33 MiB | 2.28 MiB/s, done.
Resolving deltas: 100% (2973/2973), completed with 883 local objects.
From https://github.com/openwrt/openwrt
   d6eebaf995..8a77adb048  master        -> origin/master
   09a35619b8..09dae4feac  openwrt-21.02 -> origin/openwrt-21.02
   618ab576be..506432a783  openwrt-22.03 -> origin/openwrt-22.03
 * [new tag]               v22.03.0-rc5  -> v22.03.0-rc5
M	include/kernel-version.mk
M	include/netfilter.mk
M	include/package-ipkg.mk
M	include/target.mk
M	package/base-files/files/bin/config_generate
M	package/base-files/files/etc/banner
M	package/base-files/files/etc/board.d/99-default_network
M	package/base-files/files/sbin/sysupgrade
M	package/boot/arm-trusted-firmware-mediatek/Makefile
M	package/boot/uboot-envtools/Makefile
M	package/boot/uboot-mediatek/Makefile
D	package/boot/uboot-mediatek/patches/002-nand-add-spi-nand-driver.patch
D	package/boot/uboot-mediatek/patches/003-mt7622-uboot-add-dts-and-config-for-spi-nand.patch
D	package/boot/uboot-mediatek/patches/004-configs-enable-mtd-and-mtk_spi_nand-in-defconfig.patch
D	package/boot/uboot-mediatek/patches/005-update-bpir2-defconfig.patch
D	package/boot/uboot-mediatek/patches/010-no-binman.patch
D	package/boot/uboot-mediatek/patches/010-update-u7623-defconfig.patch
M	package/boot/uboot-rockchip/Makefile
M	package/boot/uboot-rockchip/patches/002-spl-remove-dtoc-of-pdata-generation.patch
D	package/boot/uboot-rockchip/patches/100-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch
M	package/boot/uboot-rockchip/src/of-platdata/nanopi-r2s-rk3328/dt-structs-gen.h
M	package/firmware/ipq-wifi/Makefile
M	package/kernel/linux/modules/crypto.mk
M	package/kernel/linux/modules/fs.mk
M	package/kernel/linux/modules/input.mk
M	package/kernel/linux/modules/netfilter.mk
M	package/kernel/linux/modules/netsupport.mk
M	package/kernel/linux/modules/other.mk
M	package/kernel/linux/modules/sound.mk
M	package/kernel/linux/modules/usb.mk
M	package/kernel/rtl8812au-ct/Makefile
D	package/kernel/rtl8812au-ct/patches/003-wireless-5.8.patch
D	package/network/config/netifd/Makefile
D	package/network/config/netifd/files/etc/hotplug.d/iface/00-netstate
D	package/network/config/netifd/files/etc/hotplug.d/net/20-smp-packet-steering
D	package/network/config/netifd/files/etc/init.d/network
D	package/network/config/netifd/files/etc/uci-defaults/14_migrate-dhcp-release
D	package/network/config/netifd/files/etc/udhcpc.user
D	package/network/config/netifd/files/lib/netifd/dhcp.script
D	package/network/config/netifd/files/lib/netifd/proto/dhcp.sh
D	package/network/config/netifd/files/lib/network/config.sh
D	package/network/config/netifd/files/sbin/devstatus
D	package/network/config/netifd/files/sbin/ifdown
D	package/network/config/netifd/files/sbin/ifstatus
D	package/network/config/netifd/files/sbin/ifup
D	package/network/config/netifd/files/usr/share/udhcpc/default.script
M	package/network/ipv6/6in4/Makefile
M	package/network/ipv6/6in4/files/6in4.sh
M	package/network/services/dnsmasq/files/dnsmasq.init
M	package/system/opkg/files/customfeeds.conf
M	scripts/download.pl
M	target/linux/bcm27xx/Makefile
M	target/linux/bcm27xx/image/cmdline.txt
D	target/linux/bcm27xx/patches-5.4/950-1031-net-lan78xx-Ack-pending-PHY-ints-when-resetting.patch
[... 768 lines cut ...]
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/5.4/luci/protocols/luci-proto-ppp compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-text clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-text compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-sys clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/golang/golang-golang-x-sys compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/protobuf clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/protobuf compile
 make[3] -C package/libs/wolfssl clean-build
 make[3] -C package/libs/wolfssl compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/iperf3 clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/iperf3 compile
 make[3] -C package/devel/binutils clean-build
 make[3] -C package/devel/binutils compile
 make[3] -C package/libs/elfutils clean-build
 make[3] -C package/libs/elfutils compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/luci-app-iperf clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/openmptcprouter/luci-app-iperf compile
 make[3] -C package/libs/nettle clean-build
 make[3] -C package/libs/nettle compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/5.4/packages/libs/libxml2 clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/feeds/5.4/packages/libs/libxml2 compile
    ERROR: package/libs/gettext-full failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/divyansh/rpi_cm4_bloom/openmptcprouter/x86/custom_build/openmptcprouter/x86_64/5.4/source/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 1

Ysurac, 2022-07-25:
Try again ```cd x86_64/source && make```

iamdivyanshverma, 2022-07-25:
divyansh@divyansh-Latitude-7480:~/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/rpi4/5.4/source$ make
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-ledtrig-audio', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[3] diffconfig buildversion feedsversion
 make[2] package/cleanup
 make[2] package/compile
 make[3] -C package/libs/libjson-c host-compile
 make[3] -C package/libs/libubox host-compile
 make[3] -C package/system/opkg host-compile
 make[3] -C package/libs/toolchain compile
 make[3] -C package/libs/libnl-tiny clean-build
 make[3] -C package/libs/libnl-tiny compile
 make[3] -C package/libs/libjson-c clean-build
 make[3] -C package/libs/libjson-c compile
 make[3] -C package/utils/lua clean-build
 make[3] -C package/utils/lua compile
 make[3] -C package/libs/libubox clean-build
 make[3] -C package/libs/libubox compile
 make[3] -C package/system/ubus clean-build
 make[3] -C package/system/ubus compile
 make[3] -C package/system/uci clean-build
 make[3] -C package/system/uci compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/openmptcprouter/netifd clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/openmptcprouter/netifd compile
 make[3] -C package/firmware/linux-firmware clean-build
 make[3] -C package/firmware/linux-firmware compile
 make[3] -C package/firmware/prism54-firmware clean-build
 make[3] -C package/firmware/prism54-firmware compile
 make[3] -C package/kernel/linux compile
 make[3] -C package/system/ubox clean-build
 make[3] -C package/system/ubox compile
 make[3] -C package/libs/ncurses host-compile
 make[3] -C package/libs/zlib clean-build
 make[3] -C package/libs/zlib compile
 make[3] -C package/libs/ncurses clean-build
 make[3] -C package/libs/ncurses compile
 make[3] -C package/utils/util-linux clean-build
 make[3] -C package/utils/util-linux compile
 make[3] -C package/system/fstools clean-build
 make[3] -C package/system/fstools compile
 make[3] -C package/system/fwtool host-compile
 make[3] -C package/system/fwtool clean-build
 make[3] -C package/system/fwtool compile
 make[3] -C package/system/usign host-compile
 make[3] -C package/system/ucert host-compile
 make[3] -C package/utils/jsonfilter clean-build
 make[3] -C package/utils/jsonfilter compile
 make[3] -C package/system/openwrt-keyring clean-build
 make[3] -C package/system/openwrt-keyring compile
 make[3] -C package/system/usign clean-build
 make[3] -C package/system/usign compile
 make[3] -C package/base-files compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/luci/modules/luci-base host-compile
 make[3] -C package/utils/lua host-compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/luci/libs/luci-lib-ip compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/luci/libs/luci-lib-jsonc compile
[... 102 lines cut ...]
 make[3] -C package/network/config/gre compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/luci/protocols/luci-proto-gre compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/luci/protocols/luci-proto-ipv6 compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/packages/utils/dbus clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/packages/utils/dbus compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/packages/utils/attr clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/packages/utils/attr compile
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/packages/libs/libffi clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/5.4/packages/libs/libffi compile
[... cut ...]
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/openmptcprouter/libmbim clean-build
 make[3] -C /home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/feeds/openmptcprouter/libmbim compile
    ERROR: package/feeds/openmptcprouter/libmbim failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/divyansh/rpi_cm4_bloom/openmptcprouter/custom_build/arm/openmptcprouter/rpi4/5.4/source/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 1

Ysurac, 2022-07-25:
Try again, if still no success as indicated do a ```make -j1 V=s``` in source>/x86_64 directory

closed 2022-07-26 by iamdivyanshverma