Ysurac/openmptcprouter #2438: Just checking if it is Possible to connect 2 devices with one server..?

iamdivyanshverma, 2022-07-26:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2022-07-27:
Duplicate of https://github.com/Ysurac/openmptcprouter/issues/1058
In short answer: no and this is not needed in most use case. Only advantage is when you want to do some clients to clients communications.

closed 2022-07-27 by Ysurac

squeaky-geek, 2022-12-05:
i have done it, as of 12-5-2022 dual client to same vps is functioning...

iamdivyanshverma, 2022-12-05:
Hi @squeaky-geek ....
Both of your client device has different tunnel ip..?

squeaky-geek, 2022-12-05:
I didn't bother to check that, here is what I have:
Client 1: latest omr , [ server IP ] , omr server admin key. I checked the box under additional options " force retrieve all keys from server "

Client 2 : exact same config as " client 1 ". 

As far as I am aware, if there WERE an IP address conflict, neither client would be able to connect... I'll re do the double client setup and see if the ip's are different.  Side note: a webfig ( think Luci style ) for the server at least for stats would be awesome...

iamdivyanshverma, 2022-12-05:
@squeaky-geek thanks man..
Let me know if you get different ip for your device..!

Ysurac, 2022-12-05:
Glorytun doesn't allow multiple connections, so this can't work correctly this way.
You can also have multiples configuration conflicts.

rdmitry0911, 2025-03-13:
> Yes, confirmed not working :( I also would like this functionality...

Still not working, would like to make it work too