Ysurac/openmptcprouter #2441: download page returning a 404

kevinh-csalabs, 2022-07-27:
## Expected Behavior
Visiting https://www.openmptcprouter.com/download should result in a list of available downloads
## Current Behavior
Returns 404

## Specifications
N/A

Ysurac, 2022-07-28:
The page is working for me, may be a temporary problem with the CDN.

kevinh-csalabs, 2022-07-28:
I can confirm that is the case, but that the problem is still ongoing and fairly extensive.

Connections with egress via ATL and MIA fail from multiple providers (AT&T, Cloudflare, Starlink, GRUCom)
to a-us00.kxcdn.com [68.70.205.2] and go through m247 -> 37.120.220.23 which seems to be in in Europe somewhere and doesn't seem to have a route advertisement outside m247

Connection through LAX work (Spectrum enterprise)
a-us00.kxcdn.com [68.70.205.4] and go through twelve99.net -> 68.70.199.61 and work fine

Ysurac, 2022-07-28:
Another CDN is now used for download.openmptcprouter.com. Should be better.

kevinh-csalabs, 2022-07-28:
Confirmed that loads as expected. Thank you!

closed 2022-07-28 by kevinh-csalabs