Ysurac/openmptcprouter #2442: OMR-Bypass an IP address is not working

GrantSparks, 2022-07-29:
## Expected Behavior
An IP address added to OMR-Bypass should bypass the MPTCP vpn and route via a WAN connection instead of through the VPS

## Current Behavior
An IP address added to OMR-Bypass is still routing via the VPS server.  Also, I've added the MAC address of the client also to OMR bypass too so both of these settings are not working as expected.

The client is a DHCP & DNS client of the OpenMPTCProuter.lan host.  The test has been a traceroute from the client to the bypassed IP address.

I'm really just asking for help on what the next thing to check is.  Logfiles etc don't seem to contain anything unusual.  Since I'm not sure how OMR-Bypass works I don't know the best way to investigate it in the SSH shell on the router either.

## Specifications

  - OpenMPTCProuter version: v0.58.5
  - OpenMPTCProuter VPS version:  0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider:  Amazon AWS EC2
  - OpenMPTCProuter platform: RPI4

Ysurac, 2022-07-29:
What do you have via SSH on the router in ```uci show omr-bypass```, ```ip rule show``` and in  ```iptables-save``` ?

GrantSparks, 2022-07-29:
I actually have a considerable number of bypasses.  None of these are working as far as I can tell.  The test case that we are working through is the bypass of 203.53.47.17

#### root@OpenMPTCProuter:~# uci show omr-bypass

~~~
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.lo=interface
omr-bypass.lo.id='5'
omr-bypass.eth0=interface
omr-bypass.eth0.id='6'
omr-bypass.wan1=interface
omr-bypass.wan1.id='3'
omr-bypass.wan2=interface
omr-bypass.wan2.id='4'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1200'
omr-bypass.eth1=interface
omr-bypass.eth1.id='4'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='revolutiontt.me'
omr-bypass.wlan0=interface
omr-bypass.wlan0.id='10'
omr-bypass.@dpis[0]=dpis
omr-bypass.@dpis[0].proto='amazonvideo'
omr-bypass.@dpis[1]=dpis
omr-bypass.@dpis[1].proto='disneyplus'
omr-bypass.@dpis[2]=dpis
omr-bypass.@dpis[2].proto='googlemaps'
omr-bypass.@dpis[3]=dpis
omr-bypass.@dpis[3].proto='hulu'
omr-bypass.@dpis[4]=dpis
omr-bypass.@dpis[4].proto='netflix'
omr-bypass.@dpis[5]=dpis
omr-bypass.@dpis[5].proto='nintendo'
omr-bypass.radio0_network1=interface
omr-bypass.radio0_network1.id='11'
omr-bypass.wan3=interface
omr-bypass.wan3.id='10'
omr-bypass.wgOptus=interface
omr-bypass.wgOptus.id='10'
omr-bypass.usb0=interface
omr-bypass.usb0.id='8'
omr-bypass.@dpis[6]=dpis
omr-bypass.@dpis[6].proto='facebook'
omr-bypass.@asns[0]=asns
omr-bypass.@asns[0].asn='2906'
omr-bypass.@asns[0].note='Netflix'
omr-bypass.@asns[1]=asns
omr-bypass.@asns[1].asn='40027'
omr-bypass.@asns[1].note='Netflix'
omr-bypass.@asns[2]=asns
omr-bypass.@asns[2].asn='55095'
omr-bypass.@asns[2].note='Netflix'
omr-bypass.@asns[3]=asns
omr-bypass.@asns[3].asn='63293'
omr-bypass.@asns[3].note='Facebook'
omr-bypass.@asns[4]=asns
omr-bypass.@asns[4].asn='32934'
[... 440 lines cut ...]
-A zone_wan_dest_ACCEPT -o wlan0 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o usb0 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o wlan0 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain" -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHCP-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allow-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i wan1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i usb0 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i wlan0 -m comment --comment "!fw3" -j reject
COMMIT
# Completed on Fri Jul 29 07:28:55 2022
~~~

Ysurac, 2022-07-29:
I would also need the result of ```omr_dst_bypass_eth1```

GrantSparks, 2022-07-29:
Do you mean from iptables?  I'm not sure how to get that.

Ysurac, 2022-07-29:
Sorry, I mean ```ipset list  omr_dst_bypass_eth1```

GrantSparks, 2022-07-29:
Thank you.

#### root@OpenMPTCProuter:~# ipset list omr_dst_bypass_eth1
~~~
Name: omr_dst_bypass_eth1
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 520
References: 5
Number of entries: 1
Members:
203.53.47.17
~~~

kevinh-csalabs, 2022-08-24:
@Ysurac any update on this? I'm seeing the same behavior across the board for omr-bypass with the 0.59 release. I can get dns entries for these now but traceroute shows they aren't bypassing the VPS.

Ysurac, 2022-08-25:
@kevinh-csalabs Seems to be working on 0.59. Are you using as proxy shadowsocks (default) or v2ray ? and glorytun-tcp as VPN (default) ?

kevinh-csalabs, 2022-08-25:
@Ysurac I was using v2ray + glorytun-tcp but tried switching to shadowsocks + glorytun-tcp with the same result.

Example: delta.com blocks VPS ip range, so excluding delta.com and running traceroute:

Actual:
  1    <1 ms     1 ms    <1 ms  192.168.172.1
  2     1 ms     1 ms    <1 ms  OpenMPTCProuter.lan [192.168.x.x]
  3    74 ms    70 ms    87 ms  10.255.255.1
  4    80 ms    61 ms    70 ms  45.61.x.x (VPS)
  5    82 ms    75 ms    64 ms  ae23-205.cr7-mia1.ip4.gtt.net [76.74.x.x]
  6    66 ms    79 ms    79 ms  ae6.cr9-mia1.ip4.gtt.net [213.200.113.205]
  7    80 ms   136 ms    65 ms  ip4.gtt.net [98.124.172.178]
  8    69 ms    62 ms    61 ms  204.74.99.103

Expected (connected directly to default master):
  1     1 ms    <1 ms    <1 ms  StarlinkRouter.lan [192.168.x.x]
  2    54 ms    49 ms    35 ms  100.64.0.1
  3   305 ms   206 ms   200 ms  172.16.x.x
  4    39 ms    43 ms    37 ms  149.19.108.85
  5    95 ms    63 ms    53 ms  atl-b2-link.ip.twelve99.net [62.115.146.54]
  6   131 ms     *       95 ms  atl-bb1-link.ip.twelve99.net [62.115.140.6]
  7    62 ms    45 ms    98 ms  rest-bb1-link.ip.twelve99.net [62.115.138.70]
  8     *       62 ms    58 ms  204.74.99.103

Ysurac, 2022-08-25:
It's working with TCP only, I will fix that.

kevinh-csalabs, 2022-08-25:
@Ysurac Interestingly this brings up an interesting related issue.... I confirmed you are right by running a tcp trace, but I'm still getting blocked on www.delta.com (they are using akamai) despite the bypass, but if I connect directly to the master connection it works, so they are using some other method of ip detection. I noticed that if I create a bypass for akamai.com and www.akamai.com and visit https://www.akamai.com/us/en/clientrep-lookup/ I still see the VPS address. Any thoughts on how to get them to see the master interface instead? It looks like they are using google or some other service to identify the ip and comparing that to what they see directly.

kevinh-csalabs, 2022-08-25:
@Ysurac I haven't narrowed down what set is needed, but some combination of akamai and google and gstatic domains allows that check to return the master connection's ip, but this still hasn't resolved delta.com which just closes the connection immediate when connecting via openmtcprouter. I'm kind of at a loss on how they would even be able to tell the difference between the two connections.

Ysurac, 2022-08-25:
@kevinh-csalabs it's not the same problem. To bypass *.delta.com you only need to set delta.com as domain. But to really check why you can't connect you need to use your browser developers tools to find what is the site answer and what is the domain answering.

kevinh-csalabs, 2022-08-25:
@Ysurac so it seems the issue may actually be omr-bypass routing related. I'm seeing some very strange behavior.

Here is what I'm seeing:
**normal settings on openmtcprouter:** access denied (ok this makes sense, they blocked the vps)
**website added to omr-bypass with ipv6 enabled:** if ipv6 is involved it looks to still be going through the VPS (maybe because there is no otehr ipv6 routes as direct providers have only ipv4)? 
**website added to omr-bypass ipv4 only:** with ipv6 turned off there is a connection error - nothing loads at all in browser devtools, and curl just shows it redirecting then resetting the connection but the cause is obtuse:
```
curl -k -I https://www.delta.com/
HTTP/1.1 302 Moved Temporarily
Server: AkamaiGHost
Content-Length: 0
Location: https://www.delta.com/content/www/en_US/system-unavailable1.html
Date: Thu, 25 Aug 2022 21:08:33 GMT
Connection: keep-alive
```

```
curl -k -I https://www.delta.com/content/www/en_US/system-unavailable1.html
curl: (56) Send failure: Connection was reset
```

**Trying www.google.com on omr-bypass is even more bizarre** - the routing seems to be wrong AND inconsistent:

First try:
```
tracetcp www.google.com:443

Tracing route to 172.217.3.68 [mia07s54-in-f4.1e100.net] on port 443
Over a maximum of 30 hops.
1       2 ms    2 ms    2 ms    192.168.172.1
2       Destination Reached in 4 ms. Connection established to 172.217.3.68
```
but I can still load google via browser and it shows my ip as the VPS.

Second try:

```
tracetcp www.google.com:443

Tracing route to 172.217.2.196 [mia09s02-in-f4.1e100.net] on port 443
Over a maximum of 30 hops.
1       2 ms    2 ms    2 ms    192.168.172.1
2       2 ms    2 ms    5 ms    192.168.42.1    [OpenMPTCProuter.lan]
3       4 ms    3 ms    4 ms    192.168.1.1
4       39 ms   74 ms   73 ms   100.64.0.1
5       38 ms   35 ms   35 ms   172.16.249.10
6       44 ms   46 ms   36 ms   149.19.108.83
7       44 ms   39 ms   41 ms   62.115.146.54   [atl-b2-link.ip.twelve99.net]
8       53 ms   39 ms   44 ms   4.69.219.146
9       35 ms   35 ms   46 ms   142.250.165.178
10      32 ms   40 ms   43 ms   108.170.249.35
11      54 ms   30 ms   63 ms   142.251.51.16
12      65 ms   44 ms   86 ms   142.251.51.23
13      111 ms  75 ms   73 ms   216.239.54.70
14      58 ms   78 ms   49 ms   108.170.253.1
15      57 ms   51 ms   78 ms   216.239.50.109
16      Destination Reached in 58 ms. Connection established to 172.217.2.196
```

This time I can't load the page and get 
```
curl -k -I --retry-all-errors https://www.google.com/
curl: (56) Send failure: Connection was reset
```

kevinh-csalabs, 2022-08-26:
@Ysurac you can ignore my last comment - the last issue was due to downstream equipment not the router.
So to summarize:
-Non-TCP traffic ignores bypass
-IPv6 may be problematic if individual connections don't have IPv6

CEMEHOB, 2022-09-02:
I'm trying to pass UDP OpenVPN thru OpenMPTCPRouter, so it doesn't work. Bypass with UDP also doesn't work)
@Ysurac have you any plans to fix Bypass UDP?

Ysurac, 2022-09-02:
What release are you using ? it should work in v0.59.1

CEMEHOB, 2022-09-02:
I'm using 0.59.1 with VPS 1028. And it's doesn't work, I absolutely sure.

Ysurac, 2022-09-02:
What is the result of a traceroute to VPN IP (traceroute -I under Linux) ?

CEMEHOB, 2022-09-02:
semenov_e@SEMENOV-E-UB:~$ traceroute -I location-pub-kl--germany-frankfurt.aura-servers.com
traceroute to location-pub-kl--germany-frankfurt.aura-servers.com (178.162.198.111), 30 hops max, 60 byte packets
 1  _gateway (192.168.100.1)  4.164 ms  4.124 ms  4.116 ms
 2  192.168.8.1 (192.168.8.1)  28.049 ms  28.043 ms  28.037 ms
 3  178.162.198.111 (178.162.198.111)  91.322 ms  91.318 ms  91.311 ms

192.168.8.1 is a right interface

root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.lo=interface
omr-bypass.lo.id='5'
omr-bypass.eth0=interface
omr-bypass.eth0.id='6'
omr-bypass.wan1=interface
omr-bypass.wan1.id='3'
omr-bypass.wan2=interface
omr-bypass.wan2.id='4'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1200'
omr-bypass.wan3=interface
omr-bypass.wan3.id='8'
omr-bypass.wan4=interface
omr-bypass.wan4.id='9'
omr-bypass.wan5=interface
omr-bypass.wan5.id='11'
omr-bypass.wgwan5=interface
omr-bypass.wgwan5.id='12'
omr-bypass.eth1=interface
omr-bypass.eth1.id='13'
omr-bypass.tun1=interface
omr-bypass.tun1.id='15'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='location-pub-kl--germany-frankfurt.aura-servers.com	'
omr-bypass.@domains[0].interface='eth1'

root@OpenMPTCProuter:~# ipset list omr_dst_bypass_eth1
Name: omr_dst_bypass_eth1
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 512
References: 5
Number of entries: 1
Members:
46.165.225.3
root@OpenMPTCProuter:~# traceroute -I ^C
root@OpenMPTCProuter:~# ipset list omr_dst_bypass_eth1
Name: omr_dst_bypass_eth1
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 576
References: 5
Number of entries: 2
Members:
46.165.225.3
178.162.198.111

CEMEHOB, 2022-09-02:
I see udp trafic on eth0(local) interface, but can't see on eth1(bypass) interface

Ysurac, 2022-09-02:
All seems to be ok here. Can you try a traceroute -U to check if you have same result ?
You should try a tcpdump to check why traffic is not using eth1.

CEMEHOB, 2022-09-02:
semenov_e@SEMENOV-E-UB:~$ traceroute -U location-pub-kl--germany-frankfurt.aura-servers.com
traceroute to location-pub-kl--germany-frankfurt.aura-servers.com (178.162.202.15), 30 hops max, 60 byte packets
 1  * * *
 2  * * *
 3  * * *
 4  * * *
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *

root@OpenMPTCProuter:~# tcpdump -i eth0 host 192.168.100.2 and udp
20:58:53.209550 IP 192.168.100.2.38890 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.209607 IP 192.168.100.2.36795 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.209630 IP 192.168.100.2.53458 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.209656 IP 192.168.100.2.44037 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.209678 IP 192.168.100.2.56796 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.209697 IP 192.168.100.2.34072 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210076 IP 192.168.100.2.40646 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210107 IP 192.168.100.2.51414 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210117 IP 192.168.100.2.42032 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210141 IP 192.168.100.2.36113 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210142 IP 192.168.100.2.53771 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210164 IP 192.168.100.2.54006 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210194 IP 192.168.100.2.55799 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210218 IP 192.168.100.2.37477 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210262 IP 192.168.100.2.46313 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:53.210289 IP 192.168.100.2.47861 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215321 IP 192.168.100.2.53487 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215355 IP 192.168.100.2.57699 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215411 IP 192.168.100.2.41030 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215412 IP 192.168.100.2.35675 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215435 IP 192.168.100.2.60489 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215444 IP 192.168.100.2.60925 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215457 IP 192.168.100.2.36159 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215478 IP 192.168.100.2.34665 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215491 IP 192.168.100.2.59903 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215499 IP 192.168.100.2.43516 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215514 IP 192.168.100.2.57641 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215534 IP 192.168.100.2.58132 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215569 IP 192.168.100.2.50635 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215590 IP 192.168.100.2.41955 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215612 IP 192.168.100.2.57811 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]
20:58:58.215636 IP 192.168.100.2.35575 > 178.162.202.15.53: 16449 op8 [b2&3=0x4243] [17991a] [17477q] [18505n] [19019au][|domain]

root@OpenMPTCProuter:~# tcpdump -i eth1 udp
<empty>

Ysurac, 2022-09-02:
This doesn't resolve to same IP as before. You may have a cache DNS somewhere so this ip may be not available in ipset.

CEMEHOB, 2022-09-02:
root@OpenMPTCProuter:~# ipset list omr_dst_bypass_eth1
Name: omr_dst_bypass_eth1
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 896
References: 5
Number of entries: 7
Members:
91.207.172.202
46.165.225.3
178.162.198.111
77.243.181.6
91.207.172.13
178.162.202.15
78.159.101.93

It present in ipset, sure.

Ysurac, 2022-09-02:
If possible, can you try another port than 53 UDP ?
And try a tcpdump -i any host 178.162.202.15

CEMEHOB, 2022-09-02:
semenov_e@SEMENOV-E-UB:~$ traceroute -U -p 1194 location-pub-kl--germany-frankfurt.aura-servers.com
traceroute to location-pub-kl--germany-frankfurt.aura-servers.com (37.120.129.18), 30 hops max, 60 byte packets
 1  * * *
 2  * * *
 3  * * *
 4  * * *
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *

root@OpenMPTCProuter:~# tcpdump -n -i any host 37.120.129.18
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
21:24:41.907340 IP 192.168.100.2.42488 > 37.120.129.18.1194: UDP, length 32
21:24:41.907341 IP 192.168.100.2.58315 > 37.120.129.18.1194: UDP, length 32
21:24:41.907810 IP 192.168.100.2.54610 > 37.120.129.18.1194: UDP, length 32
21:24:41.907812 IP 192.168.100.2.48274 > 37.120.129.18.1194: UDP, length 32
21:24:41.907813 IP 192.168.100.2.38691 > 37.120.129.18.1194: UDP, length 32
21:24:41.907813 IP 192.168.100.2.43833 > 37.120.129.18.1194: UDP, length 32
21:24:41.907982 IP 192.168.100.2.37801 > 37.120.129.18.1194: UDP, length 32
21:24:41.907984 IP 192.168.100.2.50136 > 37.120.129.18.1194: UDP, length 32
21:24:41.907985 IP 192.168.100.2.44402 > 37.120.129.18.1194: UDP, length 32
21:24:41.907987 IP 192.168.100.2.59119 > 37.120.129.18.1194: UDP, length 32
21:24:41.907988 IP 192.168.100.2.41531 > 37.120.129.18.1194: UDP, length 32
21:24:41.907988 IP 192.168.100.2.45654 > 37.120.129.18.1194: UDP, length 32
21:24:41.908132 IP 192.168.100.2.50930 > 37.120.129.18.1194: UDP, length 32
21:24:41.907990 IP 192.168.100.2.38394 > 37.120.129.18.1194: UDP, length 32
21:24:41.907991 IP 192.168.100.2.44939 > 37.120.129.18.1194: UDP, length 32
21:24:41.907992 IP 192.168.100.2.51194 > 37.120.129.18.1194: UDP, length 32
21:24:46.912621 IP 192.168.100.2.46779 > 37.120.129.18.1194: UDP, length 32
21:24:46.913609 IP 192.168.100.2.39608 > 37.120.129.18.1194: UDP, length 32
21:24:46.913611 IP 192.168.100.2.54762 > 37.120.129.18.1194: UDP, length 32
21:24:46.913838 IP 192.168.100.2.58240 > 37.120.129.18.1194: UDP, length 32
21:24:46.913840 IP 192.168.100.2.37033 > 37.120.129.18.1194: UDP, length 32
21:24:46.913891 IP 192.168.100.2.44184 > 37.120.129.18.1194: UDP, length 32
21:24:46.913841 IP 192.168.100.2.50692 > 37.120.129.18.1194: UDP, length 32
21:24:46.913893 IP 192.168.100.2.53652 > 37.120.129.18.1194: UDP, length 32
21:24:46.913842 IP 192.168.100.2.37248 > 37.120.129.18.1194: UDP, length 32
21:24:46.913894 IP 192.168.100.2.52425 > 37.120.129.18.1194: UDP, length 32

root@OpenMPTCProuter:~# ipset list omr_dst_bypass_eth1
Name: omr_dst_bypass_eth1
Type: hash:net
Revision: 6
Header: family inet hashsize 64 maxelem 65536
Size in memory: 640
References: 5
Number of entries: 3
Members:
46.165.225.3
146.70.36.194
37.120.129.18

Ysurac, 2022-09-02:
traceroute should give you a result... and tcpdump another result. It's like if UDP output is blocked.
Did you modify router firewall config ?

CEMEHOB, 2022-09-02:
No, just default config.

Ysurac, 2022-09-02:
Do you see something using tcpdump on the VPS side ?

CEMEHOB, 2022-09-02:
I see only DNS UDP trafic on VPS side

closed 2022-12-07 by github-actions[bot]