Ysurac/openmptcprouter #2444: Possible to connect to openmptcp vps shadow socks using software client  ??

FREEEEMANX, 2022-07-30:
## Expected Behavior
Being able to connect to shadow socks to vps with shadow socks software client on windows os.

## Current Behavior
Tried to connect using a ss client with the correct ss key but couldnt.
Its a general question, if its possible or not with some existing ss client or is your implementation not possible due to specific mods ?
Usage would be to have the bonding on home lan as normal and while roaming with a laptop to connect to ss not for the bonding part but for the ss part, and avoid installing another vpn on vps like wireguard etc...

## Specifications

  - OpenMPTCProuter version: any 
  - OpenMPTCProuter VPS version: any
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-07-30:
The shadowsocks version used on VPS is patched and in default usage it's not possible to use it with something else than OpenMPTCProuter.

closed 2022-07-30 by Ysurac

FREEEEMANX, 2022-07-30:
That s what I feared when I was testing and it didnt connect.  Out of tech curiosity, if it wasnt patched, it would worked right ? Even with two clients connected at the same time - one software ss and the openmptcp router ... ? No chance to patch a soft ss client so I can do what I needed ?

 Thanks anyway, you rock !