Ysurac/openmptcprouter #2446: Gateway Down PPPOE Connection

andreaconfa, 2022-07-31:
Gateway UP

## The gateway setted up as PPPOE is showing as DOWN but is working correctly. If i set it as DHCP or static ip all is showed as good.

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: Aruba IT
  - OpenMPTCProuter platform: x86_64

[image]

Ysurac, 2022-07-31:
Can you test 0.59rc1 https://github.com/Ysurac/openmptcprouter/issues/2077 ?

andreaconfa, 2022-07-31:
how to update? i have to update both vpu and router?

> Can you test 0.59rc1 #2077 ?

Ysurac, 2022-07-31:
Yes, update VPS and then the router using 5.4 based release.

andreaconfa, 2022-07-31:
the script is telling me that the certificate is not trusted and expired

Ysurac, 2022-07-31:
Update your VPS: ```apt-get update && apt-get upgrade```
If you still have the issue, or if you don't want to update it: ```apt-get install -y ca-certificates && sed -i '/^mozilla\/DST_Root_CA_X3/s/^/!/' /etc/ca-certificates.conf && update-ca-certificates -f```

andreaconfa, 2022-07-31:
vps updated to Version 0.1027-test 5.4.207-mptcp

How to update Router?

andreaconfa, 2022-07-31:
is possible to upload or update the files without re-flash the ssd?

Ysurac, 2022-07-31:
https://github.com/Ysurac/openmptcprouter/wiki/Router-install#update

andreaconfa, 2022-07-31:
updated bot vps and router but the issue still there

> Can you test 0.59rc1 #2077 ?

andreaconfa, 2022-07-31:
can this issue be related to configuration error?

Ysurac, 2022-07-31:
Maybe, I don't test PPPoE  I would need the result of ```uci show network``` via SSH or in System->OpenMPTCProuter, "Show all settings" tab line beginning with "network."
I also need the result of ```ip r```, ```ip a``` and ``ifstatus WAN_xxxx``` (replace xxx with the name of your WAN) via SSH.

I see the detected gateway is 192.168.100.1 when IP begin with 87.14.xxxx, maybe I detected wrongly the IP or gateway...

andreaconfa, 2022-07-31:
uci show netork:
`network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='7'
network.globals=globals
network.globals.ula_prefix='fd04:a97c:d238::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.globals.mptcp_version='0'
network.globals.congestion='bbr2'
network.globals.mptcp_rr_cwnd_limited='Y'
network.globals.mptcp_rr_num_segments='1'
network.lan=interface
network.lan.proto='static'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.metric='8'
network.lan.label='LAN'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.ipaddr='192.168.110.1'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.device='eth1'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.addlatency='0'
network.wan1.ipv6='0'
network.wan1.metric='3'
network.wan1.peerdns='0'
network.wan1.pppd_options='persist maxfail 0'
network.wan1.proto='pppoe'
network.wan1.username='aliceadsl'
network.wan1.password='aliceadsl'
network.wan1.auth='both'
network.wan1.label='WAN_MANGIAGALLI'
network.wan1_dev=device
network.wan1_dev.name='eth1'
network.wan1_dev.txqueuelen='20'
network.wan2=interface
network.wan2.device='eth2'
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.addlatency='0'
network.wan2.ipv6='0'
network.wan2.metric='4'
[... 120 lines cut ...]
    inet 79.24.135.77 peer 192.168.100.1/32 scope global pppoe-wan1
       valid_lft forever preferred_lft forever
31: pppoe-wan2: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN group default qlen 100
    link/ppp
    inet 79.24.135.101 peer 192.168.100.1/32 scope global pppoe-wan2
       valid_lft forever preferred_lft forever
34: ifb4eth3: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 56:6d:d3:b8:83:ae brd ff:ff:ff:ff:ff:ff
35: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none
    inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::6b57:e469:f161:efb4/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
38: ifb4pppoe-wan1: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 36:90:96:46:78:9e brd ff:ff:ff:ff:ff:ff
41: ifb4pppoe-wan2: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether b6:77:41:c5:32:89 brd ff:ff:ff:ff:ff:ff
44: ifb4tun0: <BROADCAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether be:0f:19:25:90:c4 brd ff:ff:ff:ff:ff:ff
`
ifstatus WAN_xxxx:
Not working no metter whic name i put in

> all

Ysurac, 2022-07-31:
Should be ```ifstatus wan1```

andreaconfa, 2022-07-31:
> Should be `ifstatus wan1`

reply updated with the output of the commands

Ysurac, 2022-07-31:
I don't see any error here, it's only that 192.168.100.1 doesn't answer to ping. It's not a problem.
If you want to remove the warning in status page, you can check "Disable gateway ping" in System->OpenMPTCProuter, "Advanced settings" tab.

andreaconfa, 2022-07-31:
> I don't see any error here, it's only that 192.168.100.1 doesn't answer to ping. It's not a problem. If you want to remove the warning in status page, you can check "Disable gateway ping" in System->OpenMPTCProuter, "Advanced settings" tab.

that make sense because in fact 192.168.100.1 is not the ip address of the gateway router where the PPPOE connection is established

andreaconfa, 2022-07-31:
maybe is possible to change the gateway ip with  the correct one?

andreaconfa, 2022-07-31:
how i rollback to the stable release fot he vps?

Ysurac, 2022-07-31:
The gateway seems to be what is announced by the connection.
You can launch stable install script and this will rollback. But current VPS test release will be stable release really soon now.

andreaconfa, 2022-07-31:
onother question
My setup io OPENMPTCROUTER -> Udm Pro - LAN

I have a DDNS the resolve to my WAN IP and not i setted up this DDNS with the IP of the VPS to be able to connect to my network. 
This works fine but when i'm inside my lan this stop working. How can i override the hostname to my local ip inside OPENMPTCROUTER ?

Ysurac, 2022-07-31:
you can set a custom answer to a domain in Network->DHCP&DNS, "Addresses" field with for example "/myddns.domain.com/192.168.100.52"

andreaconfa, 2022-08-01:
> you can set a custom answer to a domain in Network->DHCP&DNS, "Addresses" field with for example "/myddns.domain.com/192.168.100.52"

thankyou, i was able to make it work.

I'm experiencing another problem: From my UDM Pro i have a L2TP VPN that i use to connect to my LAN from the internet. Now with OpenMptcprouter it stopped working. I opened all the port's to the UDM via the port forwarding (the port forwarding is working bacause i can reach other services hosted bihind the udm)

Thanks

andreaconfa, 2022-08-01:
Update: cannot connect to the UDM VPN also reaching it via one of the WAN ip addred without vpn tunnel to vps (other ports working fine)

closed 2022-11-04 by github-actions[bot]

n4sty3us, 2022-11-18:
@andreaconfa Hi, I also have a TIM vdsl connection, I ask you how did you manage to create a pppoe wan?  I searched everywhere but couldn't find it