Ysurac/openmptcprouter #2452: Port opening and forwarding

FabriceC01, 2022-08-06:
## Expected Behavior
I'd like to install Rustdesk on a server in my Lan.
Rustdesk need to forward some ports to permit access from Internet : [21115-21119] in TCP, 21116 in UDP too

## Current Behavior
I've try this configuration but it seems not working. No access from Internet. and I test the ports wich are closed.
[image]
idem for the UDP

I'm not an expert with LAN and I'd like some help (in french if possible). 
This is my config : 
[image]

Thanks for your answer.

Fabrice

Ysurac, 2022-08-07:
You have an error in your configuration: Same interface is used for both WAN. You need to use macvlan interface in this case.
To forward a port: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

FabriceC01, 2022-08-07:
Thanks for your quick answer. I'll try to modify when I'll be back at home in a few days. I don't want to risk this modification from Internet.
My OMR is in a VM on proxmox. Do you think I would better configurate a second (and even a third) virtual LAN card in proxmox  ? Would it also be better ton install a second physical Lan card on the server ?

Kalimeiro, 2022-08-08:
you need to use macvlan for wan interfaces.

with proxmox, one virtual lan is OK, configuration is like a Raspberry PI.

with my experience, i suggest to have VLAN for each WAN connection, because some 4G routers is too verbose on local IPV4/IPV6 and it's difficult to manage IPV6 without interfering with other WAN having IPV6 enabled, Else where i have already encoutered an insolvable IPV6 propagation on my lan network and can cause big trouble with OMR IPV6, so VLAN definitively solved many problem.

On Proxmox you can enable VLAN AWARE on your physical LAN and create an virtual lan interface already taggued to the right VLAN.

but for use VLAN you need to have a switch manageable, like TP-LINK SG1016PE

FabriceC01, 2022-08-18:
Thanks for your answers, 
It's working with MacVlan.
Best regards.
Fabrice

closed 2022-11-22 by github-actions[bot]