Ysurac/openmptcprouter #2455: any other loop bug in webui, in "network" -> "interfaces"

Kalimeiro, 2022-08-07:
## Expected Behavior
when we "save" all parameters on any "interface" inside "network" -> "interfaces", will be really save one time.

## Current Behavior
When go to menu "network" -> "interfaces", click "edit" of any "interface" and do nothing but just "save", you will see an "unsaved changes" 

it's repeatable endlessly, see screenshot.

## Specifications

  - OpenMPTCProuter version: 0.59 RC2
  - OpenMPTCProuter VPS version: 0.1017-test
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI4
  - Country: FR

[image]

[image]

Kalimeiro, 2022-08-10:
@Ysurac do you have see this bug ?

Kalimeiro, 2022-09-11:
this bug still present on OMR 0.59.1

gsxryan, 2022-12-11:
I still see this behavior in 0.59.1-5.4 on RPi4, however, i did do sysupgrade and not a full install.  Trying a fresh install next.

I see this when trying to select a new Wireless Client to connect.