Ysurac/openmptcprouter #2457: Port opening and forwarding in new build 0.59rc1 not working

iamdivyanshverma, 2022-08-08:
## Expected Behavior

## Current Behavior
hi @Ysurac , I am following your configuration given on "https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding", I am trying to forward a webserver from my laptop to the vps server , below are the settings i am following , packet from the port 7104 is reaching till the tun0 of the OMR but after that it is not forwarding to the eth0 interface....

[image]

[image]

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: Digital Ocean
  - OpenMPTCProuter platform:

Ysurac, 2022-08-08:
You are redirecting traffic to 192.168.100.1 port 7104 and a SNAT rule for traffic to 192.168.100.248, but as traffic is redirected to 192.168.100.1 the SNAT rule is not used in this case. I think you want to forward to internal address 192.168.100.248 in the forward rule.

iamdivyanshverma, 2022-08-11:
Hi @Ysurac ,
Yes, i want to forward it to internal address i.e 192.168.100.248...

closed 2022-11-14 by github-actions[bot]