Ysurac/openmptcprouter #2459: V2Ray CPU power needed?

iambenmitchell, 2022-08-08:
Hello,

I was reading in [Optimization](https://github.com/Ysurac/openmptcprouter/wiki/Optimization) and saw 'If you have a router with enough CPU power'

My question is, would a 2 core 2 thread CPU be powerful enough for this? I am thinking of buying a system based on the [Celeron  J3355](https://ark.intel.com/content/www/us/en/ark/products/95597/intel-celeron-processor-j3355-2m-cache-up-to-2-50-ghz.html)

Thanks. Ps, would 4GB ram be enough? I will be combining 3x 65mbps connections together into 195mbps.

Ysurac, 2022-08-09:
Yes, it's enough, and J3355 have AES-NI support, this use less CPU.

Kalimeiro, 2022-08-16:
@Ysurac can you tell which core and ram is needed for the VPS ?

closed 2022-11-21 by github-actions[bot]