Ysurac/openmptcprouter #2461: Strange behavior of DNS resolution (0.59rc2 and 0.1027-test)

Kalimeiro, 2022-08-10:
## Expected Behavior
When ping wikipedia.org or nslookup wikipedia.org, we need to get response from OMR and showing IPV4/IPV6 of the domain.

## Current Behavior
When ping or nslookup, we have a DNS timeout, resolution is impossible.

1 domain of 3 works.

VPS can ping and nslookup works.

We tried with another DNS forwarding in unbound conf, no difference.

## Possible Solution

Absolutely no idea

## Specifications

  - OpenMPTCProuter version: 0.59-rc2
  - OpenMPTCProuter VPS version: 0.1027-test
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI4
  - Country: FR

[image]

[image]

[image]

Kalimeiro, 2022-08-10:
for now, i use DNS over HTTPS directly on Firefox to work. But with this workaround omr-bypass not working (and it's normal because OMR DNS is bypassed by Firefox DoH)

Kalimeiro, 2022-08-10:
OK i have found the problem, it was in /etc/config/unbound, we lost half of configuration without understand how it is possible.

My original config :

```
config unbound 'ub_main'
	option add_extra_dns '0'
	option add_local_fqdn '1'
	option add_wan_fqdn '0'
	option dhcp_link 'none'
	option dns64 '0'
	option domain 'lan'
	option domain_type 'static'
	option edns_size '1232'
	option extended_stats '0'
	option hide_binddata '1'
	option interface_auto '1'
	option localservice '1'
	option manual_conf '0'
	option num_threads '1'
	option rate_limit '0'
	option rebind_localhost '0'
	option rebind_protection '1'
	option resource 'default'
	option root_age '9'
	option ttl_min '120'
	option ttl_neg_max '1000'
	option unbound_control '0'
	option validator_ntp '1'
	option verbosity '1'
	option listen_port '5353'
	option enabled '1'
	option recursion 'aggressive'
	option validator '1'
	option protocol 'mixed'
	list iface_lan 'LAN_PRIVE'
	list iface_lan 'LAN_TV'
	list iface_lan 'lan'
	list iface_wan 'wan1'
	list iface_wan 'wan2'

config zone 'auth_icann'
	option enabled '0'
	option fallback '1'
	option url_dir 'https://www.internic.net/domain/'
	option zone_type 'auth_zone'
	list server 'lax.xfr.dns.icann.org'
	list server 'iad.xfr.dns.icann.org'
	list zone_name '.'
	list zone_name 'arpa.'
	list zone_name 'in-addr.arpa.'
	list zone_name 'ip6.arpa.'

config zone 'fwd_isp'
	option enabled '0'
	option fallback '1'
	option resolv_conf '1'
	option zone_type 'forward_zone'
	list zone_name 'isp-bill.example.com.'
	list zone_name 'isp-mail.example.net.'

config zone 'fwd_google'
	option enabled '0'
	option fallback '1'
	option tls_index 'dns.google'
	option tls_upstream '1'
	option zone_type 'forward_zone'
	list server '8.8.4.4'
	list server '8.8.8.8'
	list server '2001:4860:4860::8844'
[... 433 lines cut ...]
	list server '9.9.9.10'
	list server '149.112.112.10'
	option zone_name '.'

config zone 'fwd_quad9_ecs'
	option enabled '0'
	option fallback '1'
	option tls_index 'dns11.quad9.net'
	option tls_upstream '1'
	option zone_type 'forward_zone'
	list server '9.9.9.11'
	list server '149.112.112.11'
	option zone_name '.'

config zone 'fwd_quad9_secured'
	option enabled '0'
	option fallback '1'
	option tls_index 'dns9.quad9.net'
	option tls_upstream '1'
	option zone_type 'forward_zone'
	list server '9.9.9.9'
	list server '149.112.112.9'
	option zone_name '.'

```

Kalimeiro, 2022-08-10:
ok, i found a second problem, when i have upgrade my vps 0.1027-test with the latest github push, shorewall not correctly apply all new configuration like autohelpers=no, i have check and set manually, restart shorewall, and i can now nslookup and ping correctly.

I need to wait few hours to see if this resolved definitively the problem.

Kalimeiro, 2022-08-10:
only tcinterfaces and policy files have been correctly apply with the vps script, not all others files.

Kalimeiro, 2022-08-10:
many few hours later, it's OK.

Ysurac, 2022-08-11:
I'm not able to reproduce the issue with DNS configuration.
Do you have DNS working from the router ? (with a ping google.fr via SSH on the router for example)
Do you use IPv6 or not ?

Kalimeiro, 2022-08-11:
Now, yes i can ping and nslookup directly on OMR through SSH or WEB UI, and with my computer too. But it's the configuration of shorewall which solve the issue.

i have execute the debian10 script to update my 0.1027-test and only 2 files on shorewall has been update (tcinterfaces and policy).
i have update manually all others files, restart shorewall, and instantly DNS lookup and ping works.

Kalimeiro, 2022-08-11:
[image]

Ysurac, 2022-08-11:
I found the issue in VPS script, I will solve it.

Kalimeiro, 2022-08-12:
for information, on a clean install of debian and install script VPS, it's OK.

we have the problem just when we upgrade the VPS.

closed 2022-08-12 by Kalimeiro