Ysurac/openmptcprouter #2462: Pfsense with CARP + OpenMPTCPRouter for high availability

voarsh2, 2022-08-12:
OpenMPTCPRouter LAN is using 192.168.100.1/24
DHCP disabled

Label: wan1
IPv4: 192.168.200.249/24

Label: wan2
IPv4: 192.168.8.249/24

---

Pfsense WAN is 192.168.100.135/24

Pfsense LAN is 192.168.1/24 - default gateway would be OpenMPTCPRouter LAN

Devices could get an IP from Pfsense but can't access wan 1/2 (192.168.200.1/24) despite allowing LAN to this network subnet. I don't have this problem when DHCP is enabled on OpenMPTCPRouter - can access all wan networks in the browser when just using OpenMPTCPRouter.

I wanted to have two OpenMPTCPRouter with different LAN IP's, 2 Pfsense VM's using CARP for failover (and to use OpenMPTCPRouter as gateways with monitoring).

Is there a better way to do this? OpenMPTCPRouter doesn't seem capable of this?
I know it's based on OpenWrt, I had a brief look at this: https://openwrt.org/docs/guide-user/network/high-availability
- wasn't sure if it applies.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: Version 0.1026 5.4.100-mptcp
  - OpenMPTCProuter VPS provider: Contabo
  - OpenMPTCProuter platform: x86_64)

Ysurac, 2022-08-16:
Following OpenWRT keepalived doc should work with OpenMPTCProuter, but it's not tested yet. Not sure it's the best way to do that. For now I didn't have HA need, but it's on todo list.

voarsh2, 2022-10-01:
I saw on this project that HA is on the roadmap, while it is not a priority for yourself, most routers software like OpenPFSense/Pfsense have HA built into it.
I much prefer having OpenMPTCRouter with HA option than introduce Pfsense into the mix for HA.
I'll look at giving keepalived with a second instance of it.

One trouble is spinning up another instance will likely screw up DHCP/internet, I've tried changing the IP configs on the second instance before but it hasn't really worked out. Any new installation will have a default IP of 192.168.100.1, the same as my primary one.

Ysurac, 2022-10-02:
There is not problem to change default IP of OMR.
I think OpenMPTCProuter HA is not widely required for any personal usage, so it's not high priority.
And this would be more a doc than code changes.