Ysurac/openmptcprouter #2467: SQM_IFB warning in system log

pbix, 2022-08-15:
No unexpected messages in the system log.

## Current Behavior
Warnings related to SQM_IFB appear if CAKE is selected as the queuing discipline.  Also if FQ_CODEL is select an additional message appears that say that sch_fq_codel is not loaded.  Please check that fq_codel loads properly on your side or perhaps remove this as a queuing option to avoid confusion.

Excerpt from the system log below:

Sun Aug 14 21:21:24 2022 daemon.info unbound: [10882:0] info: start of service (unbound 1.15.0).
Sun Aug 14 21:21:24 2022 daemon.warn [4537]: <warn>  [SQM_IFB_15011] invalid sysfs path read for net/SQM_IFB_15011
Sun Aug 14 21:21:24 2022 daemon.warn [4537]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_15011 not found
Sun Aug 14 21:21:24 2022 daemon.warn [4537]: <warn>  [SQM_IFB_ec4d5] invalid sysfs path read for net/SQM_IFB_ec4d5
Sun Aug 14 21:21:24 2022 daemon.warn [4537]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_ec4d5 not found
Sun Aug 14 21:21:24 2022 daemon.warn [4537]: <warn>  [SQM_IFB_9530f] invalid sysfs path read for net/SQM_IFB_9530f
Sun Aug 14 21:21:24 2022 daemon.warn [4537]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_9530f not found
Sun Aug 14 21:21:24 2022 user.notice firewall: Reloading firewall due to ifup of wan1 (eth0.3)
Sun Aug 14 21:21:24 2022 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sun Aug 14 21:21:25 2022 user.notice mptcp: Reloading mptcp config due to ifup of wan1 (eth0.3)
Sun Aug 14 21:21:25 2022 daemon.warn [4537]: <warn>  [SQM_IFB_cca16] invalid sysfs path read for net/SQM_IFB_cca16
Sun Aug 14 21:21:25 2022 daemon.warn [4537]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_cca16 not found
Sun Aug 14 21:21:25 2022 daemon.warn [4537]: <warn>  [SQM_IFB_6309a] invalid sysfs path read for net/SQM_IFB_6309a
Sun Aug 14 21:21:25 2022 daemon.warn [4537]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_6309a not found
Sun Aug 14 21:21:25 2022 daemon.warn [4537]: <warn>  [SQM_IFB_d2327] invalid sysfs path read for net/SQM_IFB_d2327
Sun Aug 14 21:21:25 2022 daemon.warn [4537]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_d2327 not found

## Steps to Reproduce the Problem

  1.  Fresh load of V0.95rc3 on RPI4
  2.  Configure wan1 as eth0.2 and wan2 as eth0.3
  3.  Enable cake as the queuing disc on these two VLANs.

## Context (Environment)
Just starting with Cake so not report of operation just yet but these messages are worrying me.  This is a fresh load.  I am using VLANs for my two ISP connections (ie eth0.2 & eth0.3), perhaps this is related.

## Specifications

  - OpenMPTCProuter version: v0.59.rc3
  - OpenMPTCProuter VPS version: wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh
  - OpenMPTCProuter VPS provider: linode running Debian 10.
  - OpenMPTCProuter platform: RPi4
  - Country: USA

Ysurac, 2022-08-15:
FQ_CODEL is integrated in the kernel and not as module, it's why you get a message when the script try to load it as module. This can be ignored.
For SQM_IFB message, it's modemmanager that try to check if he need to manager them as modem but it's IFB device. You can ignore messages too, not an issue.

pbix, 2022-08-15:
Thanks for this comment Ysurac.  Just some feedback on your RC3 as I see you are trying to get to a release.

closed 2022-08-15 by pbix