Ysurac/openmptcprouter #2471: Gestion Pare-Feu et Unifi Dream Machine

TheDrLucky, 2022-08-16:
Bonjour à tous,
j'ai un NAS a la maison avec Unraid qui gère plusieurs docker comme : Nginx Proxy Manager / Jellyfin / Nextcloud / rtorrent...

Sauf que je n'arrive pas a avoir accès de l'extérieur, de ma petite expérience je suppose un souci de mauvais redirections de ports.

Mon OMPTCR est sur un Linksys WRT3200ACM et toute ma maison est gère par mon Unifi Dream Machine qui lui gère ma redirection de port.

Comment je pourrai faire un sort que OMPTCR ne gère rien et laisse tous passé pour mon UDM ? (Si cela fonctionne comme ca.)

Merci à tous pour votre aide.
[image] (Le 192.168.100.139 est l'UDM)

Kalimeiro, 2022-08-16:
Bonjour,

Tu ne peux pas.

Obliger de rediriger les ports que tu souhaite de OMR vers ton UDM et de ton UDM vers ton périphérique final.

Pour info avec ton UDM : tu auras des problèmes DNS.

TheDrLucky, 2022-08-16:
Ok, et pas d'autre moyen pour faire fonctionné cela sinon ? des problèmes de DNS comment ?

Kalimeiro, 2022-08-16:
Des problèmes DNS : le serveur DNS sera ton UDM, pas OMR, donc OMR-BYPASS ne fonctionnera pas, et comme beaucoup de site et fournisseur tel que netflix bloque volontairement les réseaux des datacenters tel que OVH (pas pour les raisons de sécurité, mais pour vendre du nordvpn et autre solution pas terrible sur les données personnelles) tu auras des problèmes avec la mise en place des BYPASS des ASN et adresses IP.

un NAS sur la 4G ne te permettra pas de faire de la redirection de port, car la 4G fonctionne sur le principe du double nat, renseigne toi.

Si tu branche avant l'UDM, pose toi l'utilité d'avoir un UDM.

closed 2022-11-21 by github-actions[bot]