Ysurac/openmptcprouter #2472: TX power limited usb card

mattodinaturas, 2022-08-17:
Hi! I have a home in a remote area and i need to connect to a public wifi which is a bit far from me (about 100mt), so i need to have the maximum power allowed by the usb wireless card, which is 2watts. I am not worried about giving interference since the other houses are far away. Can somebody help me remove this restriction? Thanks!

## Expected Behavior
usb card transmitting at maximum power (2000mW-33dBm)

## Current Behavior
TX power, **only** when connected to the AP, is limited by the latter to 20dbm (100mW)

In system logs:
kern.debug kernel: [  123.080586] wlan0: Limiting TX power to 20 (20 - 0) dBm as advertised by dc:39:6f:f1:b1:70
(dc:39:6f:f1:b1:70 is the MAC address of my AP)

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter platform: RPI3

mattodinaturas, 2022-08-17:
I can also see from luci that the power is being limited to 20dbm, although i selected 30dbm
[image]

Ysurac, 2022-08-19:
TX power limit is made to respect country law. I don't think it's really easy to bypass.
More info on OpenWRT doc: https://openwrt.org/docs/guide-user/network/wifi/transmit.power.limits

closed 2022-11-23 by github-actions[bot]