Ysurac/openmptcprouter #2473: When OMR Bypass interface is set to none there is no dns

datapharmer, 2022-08-17:
## Expected Behavior
When an application such as netflix or hulu is set under omr-bypass and interface is set to unknown/none then it is expected to use the master interface unless the master interface is down. 

## Current Behavior
When unknown or none is set as the interface there is no dns resolution for the associated domains. Setting a specific interface behaves as expected but doesn't allow for failover in the case the selected interface is down

## Possible Solution
Make sure dns still resolves if interface is none. take into consideration if interface supports ipv6 or not as bonded connection may be ipv6 enabled but omr bypass may be through a connection that is ipv4 only

## Steps to Reproduce the Problem

  1. go to omr bypass
  2. choose hulu
  3. save and apply
  4. go to client and run nslookup hulu.com and note no result
  5. go back to omr bypass entry for hulu and select an interface
  6. save and apply
  7. go to client and run nslookup hulu.com again and note there is a result as expected

## Context (Environment)
unable to have failover for bypassed applications or domains etc

## Specifications

  - OpenMPTCProuter version: v0.58.5
  - OpenMPTCProuter VPS version: 0.59beta4-5.4
  - OpenMPTCProuter VPS provider: BuyVM
  - OpenMPTCProuter platform: rpi4
  - Country: United States (set to World)

Ysurac, 2022-08-17:
What is the OpenMPTCProuter release used ? To get VPS release, you can do a "cat /etc/motd" via SSH on the VPS.

Ysurac, 2022-08-18:
Fixed in develop branch.
Will work in next release.

closed 2022-08-18 by Ysurac