Ysurac/openmptcprouter #2474: unable to connect to wireguard vpn

datapharmer, 2022-08-17:
## Expected Behavior
Wireguard VPN Connects when selected in the wizard

## Current Behavior
Shows Gateway Down

What's been tried/tested so far:

1. Tried wireguard on the other physical interface (same behavior - gateway down message)
3. Tried changing shorewall ports for wireguard on VPS as they didn't match what was listed here: https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS
4. Tried changing IP on router wireguard interface as subnet was different

## Specifications

OpenMPTCProuter version: v0.58.5
OpenMPTCProuter VPS version: 0.59beta4-5.4
OpenMPTCProuter VPS provider: BuyVM
OpenMPTCProuter platform: rpi4

datapharmer, 2022-08-17:
I tried updating to the following versions and still see the same behavior.

VPS: < OpenMPTCProuter VPS 0.1027-test >
Router: v0.59rc3-5.4 r0+16594-ce92de8c8c

In reference to IP on router wireguard interface see attached image (this and shorewall rules have been set back to out of the box values including in the attached image)
[image]

Ysurac, 2022-08-17:
For the IP of the wg* interface, no problem here.
I see you use VLAN, what is the physical interface of Cellular interface ?
Do you have something in Status->System log about wireguard ?

There is a problem on VPS, the kernel doesn't seems to be patched with MPTCP, what is the result of "uname -a" ?

Steltek, 2022-08-17:
> For the IP of the wg* interface, no problem here. I see you use VLAN, what is the physical interface of Cellular interface ? Do you have something in Status->System log about wireguard ?
> 
> There is a problem on VPS, the kernel doesn't seems to be patched with MPTCP, what is the result of "uname -a" ? Did you reboot after install ?

I recently had this issue on my BuyVM VPS (with Debian 11). The script didn't manage to change the default that Grub boots. I had to manually update /etc/default/grub to get it to boot the mptcp-enabled kernel by default.

datapharmer, 2022-08-17:
Physical interface is eth0 (connected to managed switch with VLANs for each wan connection and lan on untagged VLAN1)
Nothing mentioning wireguard in the system logs.
5.10.0-11-amd64 #1 SMP Debian 5.10.92-1 (2022-01-18) x86_64 GNU/Linux
Yes, I did reboot, so not sure why it isn't using 5.4 as expected, that might explain why I'm getting -TCPOptionMPTCPCapable on MPTCP support check which is what sent me down this road in the first place... 

Update: Fought with grub for a while. eventually `grub-mkconfig -o /boot/grub/grub.cfg` worked on Debian to actually make it update instead of pretending to. 

@Steltek is right - it seems there is an issue with the script for Debian 11.

Confirmed everything is working now!

closed 2022-08-17 by datapharmer

Steltek, 2022-08-18:
> Update: Fought with grub for a while. eventually `grub-mkconfig -o /boot/grub/grub.cfg` worked on Debian to actually make it update instead of pretending to.

Now that you mention it, I think that was my problem too. update-grub would not actually respect the default and grub-mkconfig had to be called to do it. (I didn't keep good track of what I did unfortunately.)