Ysurac/openmptcprouter #2475: Wireless cards not working

mattodinaturas, 2022-08-17:
Hi! I'm having a lot of problems with wireless interfaces on OpenMPTCPprouter running on a RaspberryPi 3 B+. The internal wireless card is recognized most of the time, however, sometimes it's not. I can usually fix this by rebooting. 
Also, i have a RT3070 Ralink wireless card, with all the drivers, correctly recognized by the kernel. Most of the time, unfortunately, it's not listed as an interface and rebooting alone has a very low chance of fixing this problem.

Please note that all the tests were done with a completely fresh install of OpenMPTCProuter. Also, i tried with stock OpenWRT and i face no issues whatsoever with any of the two wireless cards

## Expected Behavior
Both wireless cards (internal of rpi3 and external usb RT3070) to be recognized correctly and listed as interfaces in ifconfig

## Current Behavior
wireless cards are not being recognized; if a card was correctly listed before as a wireless interface, in luci now it's listed as "generic 802.11bg":
[image]

Also in ifconfig there is no trace of either interface: 
[image]

Speaking about the external RT3070 wireless card, as you can see the packages are present and it's correctly recognized by the kernel:
[image]
[image]
[image]
[image]

After a reboot, now the internal wireless card of the RPI3 is recognized correctly, but still no luck with the external Ralink card
[image]

## Steps to Reproduce the Problem

Just try to connect a Ralink RT3070 USB wireless card and reboot a few times to see also the internal wireless card of the raspberry to be not recognized anymore as a wireless interface

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter platform: RPI3
  - Country: Italy

Ysurac, 2022-08-17:
Do you use a powered USB hub ?
USB on RPI3 is really bad.

mattodinaturas, 2022-08-17:
> Do you use a powered USB hub ? USB on RPI3 is really bad.

Hi Ysurac,
yes but still no luck. Now trying with a powered usb hub:
[image]
[image]

Also, as you can see, now even the internal card is not working. Probably a reboot will fix the latter but not the ralink external usb...
I know it works because I managed to get the Ralink working sometimes but by chance. I tried to reproduce the steps i did when it worked briefly, but to no avail, unfortunately. I also tried the x86_64 version of OpenMPTCProuter on an old Lenovo Thinkpad pc but the result is the same

mattodinaturas, 2022-08-17:
> Also, as you can see, now even the internal card is not working. Probably a reboot will fix the latter but not the ralink external usb...

Well, turns out I was wrong. I tried rebooting many times but this time even the internal wireless card didn't come up (and continues to be listed as generic 802.11bg, not recognized as an interface via ifconfig). The only way now is to flash again OpenMPTCProuter. Highly unstable and close to unusable!

mattodinaturas, 2022-08-18:
Hi @Ysurac , out of curiosity I tried the 64 bit image for the RPI3 (i have been using until now the generic RPI 2-3-432bit version). Well, it seems to be all in working condition now, both interfaces are recognized correctly every time. Even another usb wifi card works now, a TL-WN722N (which didn't work before). 

Guess there is something wrong w/ the 32bit image... Hope this issue helps development and general users. I'm open to sharing logs and other files in order to help in solving the issue w/ the 32bit build

Ysurac, 2022-08-18:
I should add not to use 32bits image. There is a problem to load all modules in memory I think.
I will maybe stop compilation of this 32bits image.

mattodinaturas, 2022-08-18:
> I should add not to use 32bits image. There is a problem to load all modules in memory I think. I will maybe stop compilation of this 32bits image.

Got it. Thank you for your support

closed 2022-08-18 by mattodinaturas