Ysurac/openmptcprouter #2477: Issue Building OMR with tag v0.59rc1 for RPI 4

saurabh-benlycos, 2022-08-18:
## Expected Behavior
Trying to build OMR with tag v0.59rc1 with the following :
git clone https://github.com/Ysurac/openmptcprouter.git
cd openmptcprouter
git checkout develop
OMR_TARGET="rpi4" OMR_FEED_SRC="v0.59rc1" ./build.sh -j4

## Current Behavior
But after compilation, we get the "**v0.59rc3**" image. 

## Specifications

  - OpenMPTCProuter version: v0.59rc1
  - OpenMPTCProuter VPS version: v0.1027
  - OpenMPTCProuter VPS provider: Digital Ocean
  - OpenMPTCProuter platform: RPI4

Kalimeiro, 2022-08-18:
you can't do this.

OMR_FEED_SRC is "master" or "develop".

saurabh-benlycos, 2022-08-18:
@Kalimeiro 
Thank you for your response.
So If I want to compile it with a specific git Tag that is "v0.59rc1".Then how to do that?

Kalimeiro, 2022-08-18:
i'm not an expert with git i can't tell you the correct response.

Ysurac, 2022-08-18:
You can use a tag in OMR_FEED_SRC
To set a version you need to set "OMR_RELEASE" to what you want (else always latest tag is set)

saurabh-benlycos, 2022-08-19:
Hi @Ysurac, 
I am selecting the v0.59rc1 tag but still getting the v0.59rc3 build. Can you please tell me the correct steps?

**************************************

benlycos@benlycos-HP-Z640-Workstation:~/divyansh/custom_build/bloom/openmptcprouter$ git checkout remotes/origin/develop
Previous HEAD position was 606f87cc Update MPTCP patch
HEAD is now at eabc121e Update MPTCP
benlycos@benlycos-HP-Z640-Workstation:~/divyansh/custom_build/bloom/openmptcprouter$ git branch -a
* (HEAD detached at origin/develop)
  develop
  remotes/origin/HEAD -> origin/develop
  remotes/origin/develop
  remotes/origin/master

benlycos@benlycos-HP-Z640-Workstation:~/divyansh/custom_build/bloom/openmptcprouter$ OMR_TARGET="rpi4" OMR_FEED_SRC="v0.59rc1" ./build.sh
M	include/kernel-version.mk
M	include/netfilter.mk
M	include/package-ipkg.mk
M	include/target.mk
M	package/base-files/files/bin/config_generate
M	package/base-files/files/etc/banner
M	package/base-files/files/etc/board.d/99-default_network
M	package/base-files/files/sbin/sysupgrade
M	package/boot/arm-trusted-firmware-mediatek/Makefile
M	package/boot/uboot-envtools/Makefile
M	package/boot/uboot-mediatek/Makefile
D	package/boot/uboot-mediatek/patches/002-nand-add-spi-nand-driver.patch
D	package/boot/uboot-mediatek/patches/003-mt7622-uboot-add-dts-and-config-for-spi-nand.patch
D	package/boot/uboot-mediatek/patches/004-configs-enable-mtd-and-mtk_spi_nand-in-defconfig.patch
D	package/boot/uboot-mediatek/patches/005-update-bpir2-defconfig.patch
D	package/boot/uboot-mediatek/patches/010-no-binman.patch
D	package/boot/uboot-mediatek/patches/010-update-u7623-defconfig.patch
M	package/boot/uboot-rockchip/Makefile
M	package/boot/uboot-rockchip/patches/002-spl-remove-dtoc-of-pdata-generation.patch
D	package/boot/uboot-rockchip/patches/100-rockchip-rk3328-Add-support-for-FriendlyARM-NanoPi-R.patch
M	package/boot/uboot-rockchip/src/of-platdata/nanopi-r2s-rk3328/dt-structs-gen.h
M	package/firmware/ipq-wifi/Makefile
M	package/kernel/linux/modules/crypto.mk
M	package/kernel/linux/modules/fs.mk
M	package/kernel/linux/modules/input.mk
M	package/kernel/linux/modules/netfilter.mk
M	package/kernel/linux/modules/netsupport.mk
M	package/kernel/linux/modules/other.mk
M	package/kernel/linux/modules/sound.mk
M	package/kernel/linux/modules/usb.mk
M	package/kernel/rtl8812au-ct/Makefile
D	package/kernel/rtl8812au-ct/patches/003-wireless-5.8.patch
D	package/network/config/netifd/Makefile
D	package/network/config/netifd/files/etc/hotplug.d/iface/00-netstate
D	package/network/config/netifd/files/etc/hotplug.d/net/20-smp-packet-steering
D	package/network/config/netifd/files/etc/init.d/network
D	package/network/config/netifd/files/etc/uci-defaults/14_migrate-dhcp-release
D	package/network/config/netifd/files/etc/udhcpc.user
D	package/network/config/netifd/files/lib/netifd/dhcp.script
D	package/network/config/netifd/files/lib/netifd/proto/dhcp.sh
D	package/network/config/netifd/files/lib/network/config.sh
D	package/network/config/netifd/files/sbin/devstatus
D	package/network/config/netifd/files/sbin/ifdown
D	package/network/config/netifd/files/sbin/ifstatus
D	package/network/config/netifd/files/sbin/ifup
D	package/network/config/netifd/files/usr/share/udhcpc/default.script
M	package/network/ipv6/6in4/Makefile
M	package/network/ipv6/6in4/files/6in4.sh
M	package/network/services/dnsmasq/files/dnsmasq.init
M	package/system/opkg/files/customfeeds.conf
M	scripts/download.pl
M	target/linux/bcm27xx/Makefile
M	target/linux/bcm27xx/image/cmdline.txt
D	target/linux/bcm27xx/patches-5.4/950-1031-net-lan78xx-Ack-pending-PHY-ints-when-resetting.patch
[... 1139 lines cut ...]
 make[3] -C package/utils/busybox compile
 make[3] -C package/utils/e2fsprogs host-compile
 make[3] -C package/utils/e2fsprogs clean-build
 make[3] -C package/utils/e2fsprogs compile
 make[2] package/install
 make[2] target/install
 make[3] -C target/linux install
 make[2] package/index
 make[2] json_overview_image_info
 make[2] checksum
Done

******************

benlycos@benlycos-HP-Z640-Workstation:~/divyansh/custom_build/bloom/openmptcprouter$ ls rpi4/5.4/source/bin/targets/bcm27xx/bcm2711/
config.buildinfo
feeds.buildinfo
openmptcprouter-v0.59rc3-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz
openmptcprouter-v0.59rc3-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-ext4-sysupgrade.img.gz
openmptcprouter-v0.59rc3-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4.manifest
openmptcprouter-v0.59rc3-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.gz
openmptcprouter-v0.59rc3-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2711-rpi-4-squashfs-sysupgrade.img.gz
packages
sha256sums
version.buildinfo

Ysurac, 2022-08-19:
Still: "To set a version you need to set "OMR_RELEASE" to what you want (else always latest tag is set)"

saurabh-benlycos, 2022-08-19:
Hi @Ysurac,
can you please tell me where to mention "OMR_RELEASE"...?

Ysurac, 2022-08-19:
Same way as OMR_TARGET, OMR_FEED_SRC. On the command line.

saurabh-benlycos, 2022-08-19:
Okay that means something like this..!!

OMR_TARGET="rpi4" OMR_FEED_SRC="v0.59rc1" OMR_RELEASE="v0.59rc1" ./build.sh -j4

saurabh-benlycos, 2022-08-19:
one more thing I wanted to ask is that here (https://download.openmptcprouter.com/release/) you have maintaining release versions but there I am not able to find v0.59rc3...?

closed 2022-11-22 by github-actions[bot]