Ysurac/openmptcprouter #2482: Linode Internet Speed limited by VPS Script

pbix, 2022-08-20:
After installing the VPS script there should not be any significant reduction in internet access speeds on Linode VPS.

## Current Behavior
I document a 2X speed reduction after installing the VPS script below.

## Possible Solution
Unknown

## Steps to Reproduce the Problem
  1. Create a two new Debian 10 Linode VPS, 1GB Shared CPU. 
  2. Install OMR VPS Script on one VPS
  3. Install and run Ookla SpeedTest [https://www.speedtest.net/apps/cli](url) on both VPS
  4. Run SpeedTest on both VPS and compare

## Context (Environment)
This issue was explored in a previous issue.  I am not sure if the issue is back or was not actually solved previously.
[https://github.com/Ysurac/openmptcprouter/issues/2086](url)

I checked the setting of ECN and it appears to be "2" after the script install on the VPS and "0" on the RPI4 router.

## Specifications

  - OpenMPTCProuter version: v0.59 rc3
  - OpenMPTCProuter VPS version: wget -O - http://www.openmptcprouter.com/server-test/debian10-x86_64.sh | sh
  - OpenMPTCProuter VPS provider: Linode Newark NJ Center
  - OpenMPTCProuter platform: RPI4
  - Country: USA

VPS with no script installed.
[image]

VPS after script install
[image]

pbix, 2022-08-20:
Adding more speed test results from my local router showing that speed to the internet is seriously limited compared to speed to the Linode VPS itself.  I am seeing my download speeds reduced to 25% of the value I can get from the VPS itself.  It suppose it is related to the above reported issue.

[image]

I do not see any records in the system log or the kernal log on my router that look suspicious to me.

My status screen on the router is all green.  I cannot see any indication of anything wrong.

pbix, 2022-08-20:
VPS CPU Usage seems to peak at around 85% when testing from the VPS itself.  These are Gbps speeds so that makes sense to me.
[image]

VPS CPU usage is around 35% when testing from the local router at Mbps speeds so it cannot explain the speed reduction.
[image]

Ysurac, 2022-08-20:
I don't understand your tests: you have speed issue on the VPS and you test speed from the router part ?

pbix, 2022-08-20:
I show that at each step performance is being reduced
VPS alone allows 4.6Gbps to internet
After script VPS is 2.2Gbps to the internet.
At the router iperf shows 26Mbps to VPS
Finally omr-speedtest shows only 4Mbps is finally achieved to the internet.

Especially the last reduction is a mystery to me.  Why is omr-speedtest so much lower than omr-iperf?

Ysurac, 2022-08-20:
Use "omr-test-speed" to really test aggregated speed on the router, and "omr-test-speed wan1" to test wan1 speed.
Same on VPS side, you can use "omr-test-speed" to test VPS speed.
"omr-speedtest" doesn't give good result, because it's only a bad test.

pbix, 2022-08-20:
I really appreciate your support Ysurac.  I have been struggling for 6 months to understand why OMR is so unstable in my environment.  I really could benefit from this technology if I could make it work.

I need to understand this your diagnostic tools better and have assumed the following:
omr-iperf is the same as iperf to the VPS but using aggregation?
omr-speedtest is the same as Ookla speedtest?
omr-test-speed is a custom test you made?
Can you correct my understanding above?

Because of your comment I tried omr-test-speed but the results are very similar to omr-speedtest.  I am only achieving 25% of the what I can achieve with omr-iperf.

[image]

pbix, 2022-08-20:
And individual tests look like this:

[image]

Ysurac, 2022-08-20:
omr-speedtest should use Ookla speedtest but it's not really accurate for MPTCP
omr-iperf use the iperf server on the VPS with an authentication
omr-test-speed use curl to download file using server with lower latency
I always recommend to test using omr-test-speed.
In your case you seem to have a problem.
All is green in status page ?
I didn't test Linode since a long time...

pbix, 2022-08-20:
Yes today everything has been green on the status page.  Also I want to mention that QoS is disabled for both links.

[image]

Ysurac, 2022-08-21:
When you go in Network->MPTCP, bandwidth tab and do a speed test, do you see multiples links used ?

pbix, 2022-08-21:
Screen shot below.   I ran omr-test-speed and then omr-iperf two times.
It is interesting that omr-iperf causes a spike in upload but not download?

[image]

Ysurac, 2022-08-21:
Try to always use "omr-test-speed" to test speed, it's a simple file download and give more accurate speed.
I'm not really sure speeds are aggregated on your graph, starlink is used (and I know there is no problem with MPTCP), but no idea for the InfinySky connection but seems to be used used (even if this seems slower). You can check if it support MPTCP using Network->MPTCP, "MPTCP support check".
It's possible that VPS speed is limited (allowing some burst but in fact quite slow) or have some shared CPU that can give random speed.

pbix, 2022-08-21:
The first result shown in my previous post (on the right side) is omr-test-speed result.  I follow that with omr-iperf so you can see the 10x slower result.  So both are already tested.

I am only able to achieve around 1-2Mbps as shown above.  It is hard to believe that Linode has any throttle at such a low value.  Why do upload speeds shown in the graph peak during iperf download tests?

I selected Linode because it appeared in the test and working VPS provider list.  Is there another provider in the USA that I can try that you know is working correctly with the VPS script?  I will switch to that one an try.

Results of MPTCP support check follow, all seems good.
[image]

[image]

Ysurac, 2022-08-21:
As indicated in the doc, VPS is not a recommendation list.
Currently on available US provider I use Vultr and DigitalOcean.
Did you try to set MPTCP over VPN on wan1 ? (Starlink is working without this, it's the only provider available worldwide that I can test :) )
omr-test-speed result are in kB/s, so you need a *8 to get in kbit/s
For omr-iperf you need to use "-R" to send the traffic from server to client (download), else you test upload speed.

pbix, 2022-08-21:
wan1 passes MPTCP support check and download traffic is on the graph.  Why do you think it is not working?

omr-iperf has no option -R?
===================================
root@MPTCProuter:/bin# omr-iperf -R
iperf3: error - unable to connect to server: Invalid argument
root@MPTCProuter:/bin# 

So now I update my understanding of these tools:
omr-iperf tests TCP upload speed to VPS using MPTCP, results in bps.
omr-speedtest is the same as Ookla speedtest, short downloads in bps
omr-test-speed is a custom test you made, down speeds in Bytes per second (kBps)
omr-test-speed-server (what is this?)

Ysurac, 2022-08-30:
for omr-iperf, you need to add VPS name as first argument, by default "vps", so "omr-iperf vps -R" should work
omr-test-speed is a script that use curl, so a simple download of a file.

ioogithub, 2022-09-29:
Hi @pbix do you have any update on this issue? Are you still seeing limited speeds with Linode?

I have been troubleshooting bandwidth issues. I switched from linode to digital ocean but I am still seeing the same issues, I can't achieve a speed with OMR that is greater than my slowest wan connection alone, let alone greater than the highest one. I don't suspect the VPS  at this point.

pbix, 2022-09-30:
I also switched to Digital Ocean and saw an improvement.  At some times I get higher that the fastest link.  But if there is any dropped packets the performance deteriorates rapidly to become worse than the lowest link.  Both of my ISPs deliver variable performance and its just the luck of the draw as to when the problems might occur.

ioogithub, 2022-09-30:
You have achieved higher than fastest link? Did you see this improvement simply be switching to Digital Ocean from Linode or did find some other setting that improved the bandwidth? You weren't getting these speeds when you opened the issue right? I am not sure that I really see any difference between Linode versus Digital Ocean to be honest, perhaps Digital Ocean is a bit faster but I am not sure.

I am currently seeing slower than the slowest link (25Mb/s 4G connection) with the bonded connection and I have upgraded to a fast x86 router with dedicated NICs so hardware isn't the bottleneck either.

In order for OMR to make sense it has to be at least as fast at the slowest link and as stable as the most stable connection but I am also seeing a bandwidth degregation  where the bonded connection gets slower and slower and doesn't recover until I reset the interfaces. Do you see a similar pattern with your starlink?

Where are you tracking dropped packets, perhaps this is the issue?

When I do an omr-test-speed, typically starlink will start out very fast 100+ for about 30 seconds then it seems to crash down to around 20-30. When it does this it crashes the 4G connection as well. I end up with 4G at 3/4 to 1/2 speed and Starlink at 3-4Mb/s. I can't seem to find what is causing this.

Are you using SQM autorotate with starlink or your other connection or can you recommend any other settings that seem to work with your setup?

closed 2023-01-04 by github-actions[bot]