Ysurac/openmptcprouter #2484: NanoPI R4S 1Gb version

Silvanskii, 2022-08-23:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

**Describe why you are asking for this support?**

**Additional context**

I have a NanoPI R4S 1Gb version.  I tried to install almost all available versions for this pc, but not one did not start. The device does not start at all. many OpenWRT images also don't work. On this version with 1 GB of RAM, only FriendlyARM version 20 was launched and the image of OpenWRT for rk33xx recommended 2021.09.13 with the file name R4S-immortalwrt-0623.zip
So I understand there is no support for the 1gb ram version of openmptcprouter yet?

Ysurac, 2022-08-24:
True, there is no support the the 1gb R4S release. It's still an available product ?

Silvanskii, 2022-08-24:
> 

yes, available, only bought 2 devices. I think it's not a problem, I'll buy others. I'm doing based on your device program
https://event-online.site/eobonding/

closed 2022-08-25 by Silvanskii