Ysurac/openmptcprouter #2485: How to upgrade?

hairy-dog, 2022-08-23:
It has been quite a long time since the last update and I don't remember the best way to do it. 
The local router (a Pi4 here) has to be updated and the VPS has to be updated.
Which should I update first? 
My guess is that first I update the VPS then backup the settings on the server, update the router, so I can restore the backup settings from the VPS, but perhaps that's not the best way.
Getting it wrong would affect my internat connection, so I'd prefer to get it right first time!

cr3ative, 2022-08-23:
I just did mine - upgrade the VPS, then upgrade the router. Take backups. For me, the VPS upgrade/install script upgraded everything without incident, I rebooted, then did a backup and sysupgrade flash to my raspberry pi, which again, went without any problems.

Ysurac, 2022-08-24:
It's the best way to update I think. First the VPS, local backup of the router then upgrade router.

flimo44, 2022-08-26:
Hello,
after each update I lose the vnstat. Is there a solution to not lose them or save them @Ysurac ?

hairy-dog, 2022-08-29:
I ended up having to update remotely, which was a bit scary. 
The VPS upgrade went fine but the reboot command didn't work, so when I tried to do the update from the router, I only got a long wait: I hadn't rebooted after I had updated the VPS, so the router update failed but disn't tell me: it just left me waiting.

In the end I realised the problem and used init6 to reboot the VPS and the router update went fine.

seijirou302, 2022-08-30:
> hello, i have upgraded mine (after reading the process of upgrade) : beginning with a backup of my router => upgrade VPS => create another VM on ESXi => restore backup in this VM => install open-vm-tools => plan a backup of this VM on my NAS => ENJOY Thanks @Ysurac for your great work : on a quand même des bons développeurs en France : cocorico <3

Have you found that the backup contains mac-address bindings?  I tried using your same process on ESXi but on restoring the backup to the new VM, none of the network interfaces worked.  I eventually figured out that the backup out of OMR contains mac-address options in the interface config.  A new VM assigns new mac addresses so the import effectively broke network access and I had no good way to fix it over console with my limited knowledge of the network stack in debian.

I got around it by unpacking the backup .tar.gz and editing the interface config files to remove the mac-address options and then repacking it in to a .tar.gz again.  Then after another new VM install and config load all was well.  Needing to do surgery on the backup file is obnoxious though.

k6TV, 2022-08-30:
Hi @seijirou302
I have launched another upgrade for VPS to 0.1028 5.4.207-mptcp and for router to 0.59.1-5.4.
I have created another VM in my ESXi for this version, and like the previous upgrade => no problem after backup restore...
Only install again open-vm-tools and again configuration for CBT in ESXi to make the hot backup everyday => nothing to report (it work great again and mac address were unchanged)

seijirou302, 2022-08-30:
Cool thanks for the feedback.  Not sure why I'm having the issue.  I'll try
generating a fresh backup and take a look.

hairy-dog, 2022-09-07:
As a footnote, because I was having other problems with the setup, I decided to revert to an older copy of the microsd. 
I put that in the RPi and started up. It was working OK, but was still on 58.1 so I used the Sysupgrade (no saving settings, no backup: this was a backup already) and it worked perfectly.
I'm not saying it isn't a good idea to back up, but it did update without a backup.

cagnulein, 2022-10-16:
newbie question: which command did you run on your VPS and on your local machine to update it?

hairy-dog, 2022-10-16:
I don't remember how I upgraded the VPS (I've done a lot of server upgrades recently, so it fades from memory. To upgrade the router I simply used the upgrade command in the router admin.

closed 2023-01-19 by github-actions[bot]