Ysurac/openmptcprouter #2487: Empty omr-admin-config.json

rantal, 2022-08-24:
## Expected Behavior

Router can contact with Server Admin Script on VPS
omr-admin working on VPS
omr-admin-config.json config is not resetting to empty state

## Current Behavior

Warning at openmptcprouter/status page:
Can't contact Server Admin Script (No result - No API Script answer)

warnings in logread on the router:
user.notice OMR-VPS: Can't get vps token, try later (can ping server vps, no server API answer)

Error on VPS syslog:
systemd[1]: Started OMR-Admin.
omr-admin.py[22368]: Traceback (most recent call last):
omr-admin.py[22368]:   File "/usr/local/bin/omr-admin.py", line 756, in <module>
omr-admin.py[22368]:     omr_config_data = json.load(f)
omr-admin.py[22368]:   File "/usr/lib/python3.7/json/__init__.py", line 296, in load
omr-admin.py[22368]:     parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
omr-admin.py[22368]:   File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
omr-admin.py[22368]:     return _default_decoder.decode(s)
omr-admin.py[22368]:   File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
omr-admin.py[22368]:     obj, end = self.raw_decode(s, idx=_w(s, 0).end())
omr-admin.py[22368]:   File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
omr-admin.py[22368]:     raise JSONDecodeError("Expecting value", s, err.value) from None
omr-admin.py[22368]: json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: omr-admin.service: Failed with result 'exit-code'.

Empty (!) file /etc/openmptcprouter-vps-admin/omr-admin-config.json

I haven't made any changes on the VPS myself, so I don't understand why omr-admin-config.json has become empty.

How I can regenerate correct omr-admin-config.json now? 

## Specifications

  - OpenMPTCProuter version:  v0.57.3 r0+15225-bfc433efd4
  - OpenMPTCProuter VPS version: 0.1024 5.4.86-mptcp
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86_64

Kalimeiro, 2022-08-25:
you need to use latest version, 0.57.3 is very old (2021-01-05)

rantal, 2022-09-08:
Updated to the latest version, but had to manually restore the omr-admin-config.json config on VPS using default template and passkey values from router config.

closed 2022-09-08 by rantal