Ysurac/openmptcprouter #2489: Specify interface for VPS to use?

iambenmitchell, 2022-08-25:
Hello,

I have found an ISP that offers L2TP service. My plan is to have the l2tp service run on a bridge on the VPS, and then have openmtcp get its internet connection from the L2TP running on the VPS. 

Essentially running the openmtcprouter through the ISP L2TP VPN in order to get a Residential IP on a digital ocean server. 

Is it possible to specify the interface for your VPS script to use?

Diagram:

eth0_VPS <-- L2TP_Interface <-- OMR_VPS_Server <-- OMR_Router <-- My home

iambenmitchell, 2022-08-25:
I'd rather run the L2TP VPN on the VPS because the ping from VPS to the ISP is 1ms, but from my house it is 23ms.

My house to VPS is 17ms so having the VPN on the VPS means it will be around 18-20ms but if I ran it from my home it would be around 40ms

Ysurac, 2022-08-26:
You have to modify ```/etc/shorewall``` configuration and default GW on your VPS.

closed 2022-11-30 by github-actions[bot]