Ysurac/openmptcprouter #2490: Speed Problems 2gbps bonding?

NiclasJohn, 2022-08-25:
## Expected Behavior
I got 2x 1gbps doqnload and want to bond it
My vps has 10gbps with 8 vcores

## Current Behavior
I only get arround 300mbps download and full upload arround 100-110mbps.
The router part runs on a ryzen 5 3400G and the vps is on not even 30% load while doing speedtest so i think theres enought headroom for more speed.
Are there settings to change for getting more speed?

## Specifications

  - OpenMPTCProuter version: v0.59-5.4-r0+16594
  - OpenMPTCProuter VPS version: 0.1027
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-08-26:
You can try to follow https://github.com/Ysurac/openmptcprouter/wiki/Optimization. You are doing speed tests from the router ? Did you test each wan speed ? https://github.com/Ysurac/openmptcprouter/wiki/Test-speed

MokkaSchnalle, 2022-10-15:
Seems to be DOCSIS Internet because of the low upload speed. Maybe full node?
Do you have full 2Gbps via normal MultiWAN load balancing?

I am getting around 1.3-1.4Gbps in MPTCP (normal 1.90Gbps). Still not good, but fine compared to single stream.
Tried all tuning stuff already.

closed 2023-01-18 by github-actions[bot]