Ysurac/openmptcprouter #2491: add adblocking support such as AdGuardHome

kevinh-csalabs, 2022-08-26:
**Is your feature request related to a problem? Please describe.**
Please consider adding an ad/malware blocking solution for dns

**Describe the solution you'd like**
Consider adding AdGuardHome 
https://github.com/AdguardTeam/AdGuardHome
or similar to the package

**Describe alternatives you've considered**
Can run pihole or other dns filter but this adds unnecessary complexity to the setup

**Additional context**
I tried passing through openmtcprouter to another device that could filter the dns further but this had other issues.

Ysurac, 2022-08-29:
This will not be installed by default, because AdGuardHome need more than 100Mo.
But it's available as package and there is a doc to install it on OpenWrt website: https://openwrt.org/docs/guide-user/services/dns/adguard-home

closed 2022-09-10 by kevinh-csalabs

yostyle, 2023-05-13:
Hi @kevinh-csalabs, @Ysurac
@kevinh-csalabs, did you follow the doc to install AdGuardHome on your Openmptcprouter instance ? On my side I tried to install it. AdGuardHome works great but my rules in OMR bypass are ignored. I guess this documentation needs to be adapted on Openmptcprouter to keep this feature.
@Ysurac could you help us about that ?