Ysurac/openmptcprouter #2493: Traffic optimalization both for TCP and UDP

pbapba512, 2022-08-26:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: V0.59
  - OpenMPTCProuter VPS version: 0.1027
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64

Its not a real issue, more over a "best practice" question... some moths ago I read the documentation, and the V2RAY was suggested to optimize both the TCP and UDP traffic.

Now the final V0.59 released, and the release notes say: V2Ray can be used only for TCP (instead of TCP & UDP). 

So now I'm confused, what are the best settings if I would like to aggregate the traffic both for the UDP and TCP? I do a lot of livestreaming via RTMP (TCP) and SRT (UDP) protocols, so I have to bond the WAN connections for TCP and UDP at the same time. 

Thank you for the suggestions!

Ysurac, 2022-08-29:
There is no change, only a checkbox to use v2ray only for TCP if you want.
And as always, the best configuration is what work the best for your usage.

ccmks, 2022-08-30:
I also realized that you can now use Shadowsocks for UDP connection if the VPN is down. Is to good to use shadowsock for UDP connection?

pbapba512, 2022-08-30:
> There is no change, only a checkbox to use v2ray only for TCP if you want. And as always, the best configuration is what work the best for your usage.

Thank you for the clarifying the things! I will test out, anyway I have a good experience with v2ray both for the TCP and UDP based livestreaming.

closed 2022-12-04 by github-actions[bot]