Ysurac/openmptcprouter #2494: VPS Script fails on omr-vps-admin version not found

JonAdamsFromNC, 2022-08-28:
## Expected Behavior
VPS script should find and install and configure dependencies automatically.

## Current Behavior
Current version of VPS script fails to find latest version of omr-vps-admin, giving message
E: Version '0.3+20220827' for 'omr-vps-admin' was not found

## Possible Solution

## Steps to Reproduce the Problem

  1. Get latest VPS script with wget
  2. Run latest VPS script

## Context (Environment)
AWS Lightsail instance
VPS buildscript pulled 2:10PM UTC -5 28/08/22

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: n/a
  - OpenMPTCProuter VPS provider: AWS Lightsail
  - OpenMPTCProuter platform: RPI4
  - Country: US

Ysurac, 2022-08-28:
I forgot to purge CDN cache of the Debian repo. Should be better now.

JonAdamsFromNC, 2022-08-28:
I had to re-create my Lightsail instance in order for it to get past that error message, something stuck in a cache.
It made it through the script, however one more error was displayed immediately before the "Success" message

E: Version '0.1028' for 'omr-server' was not found
====================================================================================
OpenMPTCProuter Server 0.1028 is now installed !

Ysurac, 2022-08-28:
Yes, it's not a problem.

closed 2022-12-03 by github-actions[bot]