Ysurac/openmptcprouter #2495: The bandwidth aggregation does not work with two 4G connections

HeroGian, 2022-08-28:
## Expected Behavior
I have two LTE connections that on average can reach about 80Mbit/s with a peak of 100Mbit/s each. I would like to aggregate the two bandwidth to reach about 160Mbit/s on average and a peak of 200Mbit/s.
I attach two screenshots of the two WANs that show the behaviour of the two connections individually. I generated the traffic using omr-test-speed eth1 and omr-test-speed eth2:

https://ibb.co/SJHVvLd
https://ibb.co/9cNFfPP

this is the network configuration:
https://ibb.co/PD9tX6r

## Current Behavior
The aggregated bandwidth is about 80Mbit/s and looking at the graph obtained using omr-test-speed, we can see that the bandwidth of the two interfaces is the half of the bandwidth that the connection should have (withoud bonding).

output of omr-test-speed:
https://ibb.co/VB9JLkK

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter VPS provider: Serverplan
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-08-28:
Update to latest release.
Check also that ISP are not sharing same resources.

HeroGian, 2022-08-29:
I updated to latest versione but with no luck. I don't think that the two connections are sharing resources.
How can I investigate the problem?
Thanks

Ysurac, 2022-08-29:
You can use "omr-test-speed wan1" and "omr-test-speed wan2" at the same time on the router, and check result on the console.
This will run speed on each wan at same time, you should get max speed on each wan. If it's not the case there is a bottleneck somewhere.

arviido, 2022-09-07:
I've got a similar issue. I'm currently running up to date versions on vps and router. This issue has persisted in atleast a few releases. I'm using 1 wifi as wan (lowest latency, master) + 2x 4G connections (both tethered from a phone).

If I use round-robin, I am able to saturate full bandwidth of all the connections, but on 'default' or 'blest', bandwidth used for one 4G connection drops off after one second or so. Somehow 'default' has aggregated all the connections on one occasion, but haven't been able to get it working after reboot and disconnecting 4G connections.

No idea how to debug further.

kevinh-csalabs, 2022-09-08:
I'm also seeing similar behavior that I've been going crazy trying to troubleshoot. In my case one connection is 5g and the other is starlink. Both connections individually have consistently double the bandwidth individually than that of the bonded connection (should be able to get 8-10MBps when bonded but actually getting 1/4 the expected speed of 1-2.5MBps).

I've confirmed proper kernel on vps, MPTCP is showing sender key for both, and everything looks good under router status.  I've tried changing schedulers and traffic control and turning on and of sqm and switching default proxy all to no avail.

I'm on the latest release .59.1-5.4 and 1028.

Ysurac, 2022-09-08:
Not sure your issues are related, and as nobody do the test I asked I don't know how this can help.

kevinh-csalabs, 2022-09-08:
@Ysurac 
connection 1:
```
Best server is http://atl.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  6 9536M    6  588M    0     0  2248k      0  1:12:22  0:04:27  1:07:55 7752k
```

Connection 2:
```
Best server is http://mia.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3 9536M    3  380M    0     0  1461k      0  1:51:23  0:04:26  1:46:57 1619k
```

Confirmed these connections don't share anything upstream via traceroute (one terminates in jax and the other in atl) and by running connections on both simultaneously with devices other than the openmptcprouter.

These speeds are extremely slow compared to connecting directly with a device other than the openmptcprouter.

Here is connected directly to connection 2 for example:
```
curl -o  /dev/null http://atl.download.datapacket.com/10000mb.bin
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 13 9536M   13 1279M    0     0  10.1M      0  0:15:43  0:02:06  0:13:37 7811k
```
It is a 10 fold difference.... seems something is wrong before we involve bonding.

Ysurac, 2022-09-08:
So this explain why OMR can give a good result if speed are bad without involving aggregation.
What is the arch used ? You use dedicated port for each connection without any switch ?

kevinh-csalabs, 2022-09-08:
Router is Pi 4B 2GB with squashfs. It does use an aruba gigabit switch; connections are segregated by vlan. Confirmed throughput on the switch is not a problem via replacing pi with laptop and testing speed.

kevinh-csalabs, 2022-09-08:
When trying to update EEPROM I get `/usr/bin/rpi-eeprom-update: line 203: syntax error: unexpected "||"`

Ysurac, 2022-09-08:
You are using v0.59.1 ? If not: update.

kevinh-csalabs, 2022-09-08:
yes, I'm on current version. I commented out 203 and 213 and was able to run it. There was an update, but same behavior after reboot. I'll try testing if vlan is the issue, but don't love the idea of relying on layer 3 for isolation between wan and lan.

kevinh-csalabs, 2022-09-09:
ok, so removing vlans from eth0 had no impact unfortunately.

Example from tonight:
Starlink: 70Mbps (direct)
5g TMobile: 180Mbps (direct)
Combined: ~60Mbps

I'm still seeing the behavior of the interfaces themselves being very slow through PI4 without any obvious cause.

I wanted to eliminate the switch and eth0 so I tested via wlan0 and still see similar speeds:

omr-test-speed wlan0
Best server is http://atl.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 9536M    2  273M    0     0  1234k      0  2:11:52  0:03:47  2:08:05 1798k

Obviously there are performance impacts with wifi on PI but this still seems unusually slow given I am easily pulling 70Mbps from this wifi connection with other devices.

bitrate for this wireless connection is: 433 Mbit/s

kevinh-csalabs, 2022-09-13:
@Ysurac I added a 3rd wan connection for troubleshooting purposes, and found the bonded speed hasn't changed significantly - it just seems to balance out over the connections, but the total aggregated speed is still less than any single wan connection alone still. I confirmed mptcp_check shows sender's key for each connection, and everything is green under the status. Any thoughts on this behavior or what I can do to diagnose the issue further?

eattil, 2022-09-13:
Postemobile is a Vodafone virtual operator. you are probably using the same resources

ioogithub, 2022-09-14:
> but the total aggregated speed is still less than any single wan connection alone still.

@kevinh-csalabs, I believe am experiencing a similar problem and came across your issue. I have been reporting bandwidth results here: https://github.com/Ysurac/openmptcprouter/discussions/2353. I have a 4G connection bonded with starlink and I always see the aggregate bandwidth slower than the faster wan connection.

I am also very new to starlink so it is hard to know what the baseline performance should be. I have noticed the speed varies considerably from moment to moment. Is this normal for starlink?

On the other hand my LTE connection is solid and consistent and always returns 24mb. I do see a bonded aggregate speed less than 24mb on occasion as well which is puzzling.

I am using an rpi 4b 8gb with two TP-UE300 USB to Ethernet dongles connected to each WAN.

I am looking for the bottleneck, if anyone has any suggestions or things I can test to find out why my aggregate bandwidth is lower than the single connection I would appreciate hearing about it.

kevinh-csalabs, 2022-09-15:
@Ysurac Ok, after some digging - it turns out some of the 5g operators such as t-mobile have major performance penalties when operating on ipv4 only, so enabling ipv6 on the interface and testing over ipv6 gives the connections the expected throughput for the individual wans. _IPv6 was off by default when configuring the interfaces on openmptcprouter which explains the discrepancy between devices. I'm not sure if there is any negative impact of enabling ipv6 by default to avoid others running into this issue, but I'll say that this was a royal pain to track down and honestly kind of bizarre._

Further, in looking at logs I was seeing a lot of wlan disconnects. I tracked this down to a regulatory mismatch between systems. I set it to us using: `iw reg set US` which resolved that problem. _It might be good in a future version to have more options than china and global in config such as us, eu etc. and utilize this to set wifi region to avoid connection issues when channels change._

Finally I was seeing a large number of sqm entries and determined that when autorate was set under openmptcprouter it was expecting values for min/max upload/download under network > SQM QoS as well which weren't being set. Filling these in manually resolved the system.log complaints for the most part, so it might be necessary to require those on the settings page if that's enabled (I see a note about min/max being used, but doesn't seem to imply they are required for avoid errors). 

I am still getting some sysfs path warnings I'm not sure what to do with and occasional broken pipe messages for sqm:
```
Wed Sep 14 14:13:19 2022 daemon.warn [5176]: <warn>  [SQM_IFB_62692] invalid sysfs path read for net/SQM_IFB_62692
Wed Sep 14 14:13:19 2022 daemon.warn [5176]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_62692 not found
Wed Sep 14 14:13:19 2022 daemon.warn [5176]: <warn>  [SQM_IFB_17baf] invalid sysfs path read for net/SQM_IFB_17baf
Wed Sep 14 14:13:19 2022 daemon.warn [5176]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_17baf not found
Wed Sep 14 14:13:19 2022 daemon.warn [5176]: <warn>  [SQM_IFB_fc096] invalid sysfs path read for net/SQM_IFB_fc096
Wed Sep 14 14:13:19 2022 daemon.warn [5176]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_fc096 not found
Wed Sep 14 14:13:19 2022 daemon.warn [5176]: <warn>  [SQM_IFB_c4727] invalid sysfs path read for net/SQM_IFB_c4727
Wed Sep 14 14:13:19 2022 daemon.warn [5176]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_c4727 not found
Wed Sep 14 14:13:20 2022 daemon.warn [5176]: <warn>  [SQM_IFB_667f3] invalid sysfs path read for net/SQM_IFB_667f3
Wed Sep 14 14:13:20 2022 daemon.warn [5176]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_667f3 not found
Wed Sep 14 14:13:20 2022 daemon.warn [5176]: <warn>  [SQM_IFB_25ffa] invalid sysfs path read for net/SQM_IFB_25ffa
Wed Sep 14 14:13:20 2022 daemon.warn [5176]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_25ffa not found
```

I also noticed that when using macvlan for wan if you enable ipv6 under advanced for openmptcprouter settings then it saves and works ipv6 on the vps and dhcp which seems to work even if wan is ipv4 but you can't change the macvlan settings after or do it in the other order because it considers ipv6 on each wan to then be required (which isn't really the case as it can establish ipv4 to the server - ipv6 isn't possible with starlink for example right now). Using vlans with dhcp doesn't require this and seems to work without issue on a pi 4b for what that's worth.

Aggregated bandwidth is much closer to what I'd expect now, so I'll continue to try some tuning.  

My remaining questions are: 

1. How is ipv6 vs ipv4 selected for server connection and is there a way to set a preference of ipv6 or ipv4 per wan for wan to vps etc? 
2. Should I be concerned about these remaining sysfs path warnings?

closed 2022-12-20 by github-actions[bot]