Ysurac/openmptcprouter #2497: qmi modem ( mc7750 ) will not initialize

squeaky-geek, 2022-08-29:
PLEASE NOTE:
server is good, default setup ( using ethernet ports and dual GX440's ) works fine.
however, when the modems are migrated into the unit ( apu 3c4 );
1: "openwrt" sees them, connects + configures, INSTANTLY... ( no mptcp though *sad face* )
2: openmptcprouter ( OMR ) ( see images please ) refuses to handle them correctly ( appropriate modules, drivers, ect. seem to be listed + enabled ) the modems BOTH show as "qmi~wwan0" + "qmi~wwan1" however, see " issue " , please reference attached images.

build:
Pc Engines APU 3c4
2x Sierra Wireless MC7750 mPcie Modems
     ^- using qmi-utils OR Modem Manager

issue:
neither modem will receive ; gateway, IP, at first ONE would connect without gateway, now... neither modem is functioning correctly.
i am unable to " Force " the original situation of only one modem partially connecting.

steps taken so far:
sim swaps, ( i have 2 sets available, all 4 sim cards work elsewhere. )
[ reboot ] , [ update + reboot ] , [ change modem init wait-time ] , [ reset / start from scratch ] , [think bad hardware? ] 
>[ power off ] , [swap ssd to " openwrt" ] , openwrt boots and shows BOTH devices connected to carrier network grrrr...

so i try a FRESH opnmptcprouter install....  nope... issue persists ( is it possible using 2 of the same model modem is causing a conflict? )

images may give some insight, if logs are needed please advise how to create them, i have ssh available.
note: onemarcfifty sent me here, ONLY because the " default OpenWRT " sees the modems, he believes it is something to do with the init script, however i am a hardware guy, not a code guy... i promise i will do my best to follow your instruction, and if a tutorial is available please link, i am accustomed to doing extensive research. ( i have never coded for modems before :/ ) -thank you.

- i have more images upon request.

[image]
[image]
[image]
[image]
[image]
[image]

Ysurac, 2022-08-29:
MAC address seems to be the same for both modems.
Can you try to set only one modem first ?

squeaky-geek, 2022-08-29:
> MAC address seems to be the same for both modems. Can you try to set only one modem first ?

I will attempt single modem install and report back.

squeaky-geek, 2022-08-30:
please see images attached:

[image]
[image]
[image]
[image]
[image]

squeaky-geek, 2022-08-30:
** both qmi AND modem manager have been tried with same error / issue

squeaky-geek, 2022-08-30:
all screen caps so far:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

squeaky-geek, 2022-08-30:
Duplicate Mac address was from the OpenWRT screenshot** OpenWRT works with DUAL lte modems just fine, however, openmptcprouter does not.

Ysurac, 2022-08-30:
Duplicate MAC can be a problem with OpenMPTCProuter.
Too many captures here... I really don't know what you are doing.
Use Network->Interfaces to configure the interface, ModemManager as protocol and if possible full path to the device (/sys/devices/platform/soc/20980000.usb/usb******) but ModemManager should directly show you device name (and will use full path). Use also IPv4 Only on IP Type. If you get gateway down in status page it's often not a problem.

squeaky-geek, 2022-08-30:
something changed xD i am using modem manager with single modem and it is connected, i will attempt dual modem now.

squeaky-geek, 2022-08-30:
see images: this is where I'm at. i dont understand the error displayed, i have not yet changed any advanced images
[image]
[image]
[image]

squeaky-geek, 2022-08-30:
[image]

squeaky-geek, 2022-08-31:
> Duplicate MAC can be a problem with OpenMPTCProuter. Too many captures here... I really don't know what you are doing. Use Network->Interfaces to configure the interface, ModemManager as protocol [...]

I have followed these instructions exactly and I now get a different error " mm teardown in progress " I have no idea what that means and google didn't help.

squeaky-geek, 2022-08-31:
well... i feel like an idiot for not doing this first:
delete interfaces ... reboot ... recreate wan1 , recreate wan2 ... reboot ... wait ... and the're up!!   modem manager for the win.
thank you sir, qmi would not work for these modems, the solution for on board Sierra Wireless MC7750 is Modem Manager

closed 2022-08-31 by squeaky-geek