Ysurac/openmptcprouter #2498: install tutorial for Banana Pi BPI-R64

Marco-LB, 2022-08-29:
Hello everybody,
i need help, how to install openmptcprouter on Banana Pi BPI-R64? what procedure do I have to do to be able to get it working?

wouam31200, 2022-09-03:
Bonjour Ysurac, j'ai tourné dans tous les sens avec des anciennes version, et en essayant de suivre le wiki, je n'y arrive pas.
Il doit bien avoir un truc que je loupe, soit le kernel, soit l'installation. Je suis perdu!
Si tu as un peux de temps et si tu peux faire une petite doc, ce serait cool
Je te remercie par avance.

Marco-LB, 2022-09-06:
a working sd card image would be enough for me

Ysurac, 2022-09-06:
You tried to use an OpenWRT image and to update it using OpenMPTCProuter image ?

Marco-LB, 2022-09-07:
yes I tried both via ssh and via web interface, but the system does not update, it remains OpenWrt

wouam31200, 2022-09-07:
same for me

wouam31200, 2022-09-09:
I managed to install: (https://snapshots.openmptcprouter.com/5.15/bpi-r64/targets/mediatek/mt7622/)

Web version: openmptcprouter v0.59-5.15 r0+20029-3c06a34
Serial port version:
PACKAGE:     openmptcprouter
VERSION:     v0.58.5
TARGET:      bpi-r64
ARCH:        aarch64_cortex-a53

Is it normal that it's not the same version?

If I update (https://download.openmptcprouter.com/release/v0.59.1/bpi-r64/targets/mediatek/mt7622/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-mediatek-mt7622-bpi_bananapi-r64-squashfs-sysupgrade.bin) it won't start!

Ysurac, 2022-09-09:
I think only release based on kernel 5.15 can work correctly.
Snapshot is 0.59 (with the bug in SSH access that display 0.58.5), I didn't build a new snapshot since some times.
This is the latest release: https://download.openmptcprouter.com/release/v0.59.1-5.15/bpi-r64/targets/mediatek/mt7622/

Kirball, 2022-09-15:
I see there was a new version for the bpi r64 added recently.  Does it work now?  Should I drag out my Bpi and try again or does it still get stuck in a boot loop?  Has anyone used it with any success?

Ysurac, 2022-09-15:
5.15 kernel based should work, but a new snapshot is compiling that should work better.

BBGScc, 2022-09-21:
Hello surac, @Ysurac 
I have a problem, I follow the method to flash and start successfully, but I can't save my modified configuration, what is the reason?

Marco-LB, 2022-10-04:
I solved it by switching platforms

closed 2022-10-04 by Marco-LB