Ysurac/openmptcprouter #2499: Reporting wrong version

obi12341, 2022-08-29:
SSH Login Banner is always reporting wrong version:
```
BusyBox v1.33.2 (2022-05-17 22:11:28 UTC) built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|
       |_|
------------------------------------------------------------------------------
   (r0+16594-ce92de8c8c)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.58.5
 TARGET:      x86_64
 ARCH:        x86_64

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Mon Aug 22 21:41:55 UTC 2022
-----------------------------------------------------
```

although I have upgraded from 0.58.5 to 0.59.0. Backend is reporting the right version (see screenshot) but also always finds newer version to upgrade to.

[image]

Ysurac, 2022-08-30:
Release if fixed in 0.59.1 for the SSH login banner.
For sysupgrade this will be in a next minor release.

hairy-dog, 2022-08-30:
I am offered upgrade to 0.59.1 but after running the sysupgrade and the automatic reboot, I am still on 0.59-5.4

hairy-dog, 2022-08-30:
[image]
It looks as if the sysupgrade is fetching the old file.

Ysurac, 2022-08-31:
It's fixed now. CDN cache not flushed for this part, so this was still old release.

fsufyan, 2022-08-31:
I am still getting this on RPI 3B+
[image]

gregorst3, 2022-09-04:
> It's fixed now. CDN cache not flushed for this part, so this was still old release.

I upgraded to 0.59.1 but sysupgrade still shows to update to 0.59.1
[image]

Ysurac, 2022-09-04:
I don't see this as a big problem, but this will be solved in next release.

gregorst3, 2022-09-04:
> I don't see this as a big problem, but this will be solved in next release.

Absolutely not a problem at all, was just to report!
Thanks fo this awesome project

ogmkp, 2022-10-22:
Juste a message to don't forget this ;)

closed 2023-01-25 by github-actions[bot]