Ysurac/openmptcprouter #2500: Really slow.

hairy-dog, 2022-08-30:
## Expected Behavior
Everything was working just fine, then something happened. I think we had a power glitch (even though it has a UPS Hat). Anyway, the OMR router, which runs on a Pi4, reported a readonly file system.

I'm away at present so I have no way to create and install a replacement SD card. I fixed the file system as well as I could, and it is no longer readonly. I tried updating to the new version both on the VPS and on the router. 

## Current Behavior
The maximum down speed is 4Mb/sec.

The current wan connections are vDSL at 39.45Mb/sec, vDSL at 31Mb/sec and 4G at (maybe 30Mb/sec).

I tried going through each of the processes in system/startup and restarting them one by one. That seemed to solve the problem. I was then getting over 40 down, 20 up. But the next day it is back to 4 down, 1.5 up.

If I use the bypass feature to connect a specified lan IP through wan1: I get about 30 down, 8 up, as I would expect from that single connection.

Any suggestions about how I can work out that is wrong and fix it? No clues in the log.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1028

  - OpenMPTCProuter VPS provider: Bitfolk, London
  - OpenMPTCProuter platform: RPi4

duckigol, 2022-08-30:
Try these settings:

**network->mptcp**
   Congestion Control: bbr2 (only if your raspberry supports 64bits)

**system->openmptcprouter**
wan(1,2,3) Enable sqm and set Download/Upload speed (Kb/s)

**network->sqm**
wan(1,2,3):
       quee discipline: cake
      Queue setup script: piece of cake

**tun0:**
       quee discipline: cake
      Queue setup script: piece of cake
      link layer adaptation: None

hairy-dog, 2022-08-30:
Sadly, that didn't seem to make an appreciable difference.

duckigol, 2022-08-30:
could it be the ethernet cable
Your micro-sd must be at least class 10

Ysurac, 2022-08-30:
Check on VPS that the correct kernel is loaded via "uname -a"
Check also that all is green in OpenMPTCProuter status page
Check speed of interfaces using "omr-test-speed wan1" for each WAN interface you have, and "omr-test-speed" to check aggregated speed (ctrl+c to stop after 2 minutes). Run also "omr-test-speed" on the VPS.

hairy-dog, 2022-08-30:
uname -a
Linux lon 5.4.207-mptcp #1 SMP Sun Jul 24 14:39:44 UTC 2022 x86_64 GNU/Linux

omr-test-speed wan1 (Vodafone vDSL)
Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 9536M    4  412M    0     0  3511k      0  0:46:21  0:02:00  0:44:21 3731k

omr-test-speed wan3 (Vodafone LTE)
Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 9536M    1  166M    0     0  1386k      0  1:57:24  0:02:02  1:55:22 1238k

omr-test-speed wan5 (this is SKY vDSL sync at 30Mb/sec)
Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:30 --:--:--     0
curl: (28) Failed to connect to lon.download.datapacket.com port 80 after 30703 ms: Operation timed out

omr-test-speed (router)
Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0 13.8M    0     0   115k      0 23:24:49  0:02:02 23:22:47 26326

omr-test-speed (VPS)
Best server is http://speedtest.tele2.net/1000GB.zip, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 1000G    1 12.8G    0     0   107M      0  2:38:49  0:02:02  2:36:47  108M

Ysurac, 2022-08-31:
Seems that you have a problem with wan5. What is the aggregated result when you disable MPTCP on this interface ?

hairy-dog, 2022-08-31:
with it disabled:
Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  5 9536M    5  501M    0     0  4176k      0  0:38:58  0:02:02  0:36:56 5277k
with it reenabled:

Best server is http://lon.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (7) Failed to connect to lon.download.datapacket.com port 80 after 2 ms: Connection refused

Enabling MPTCP over VPN for that wan makes no difference

I tried speedtest.net just now and I got 27.38 down, 5.94 up. That's slower than wan1 on its own.

Ysurac, 2022-08-31:
So there is a problem with wan5. As aggregated speed without it give wan1+wan3 speed
The "Operation timed out" is not good, and it's not MPTCP over VPN that can solve this. Check MTU and check if the connection is really working.

hairy-dog, 2022-08-31:
Wan5 is reporting sync at 30 down, but it is a strange connection. Its status says:
Traffic Type:PTM
Line Rate - Upstream (Kbps):6690
Line Rate - Downstream (Kbps):30687
and it says
Network TypeMER but I have nno idea what that means.

Is there any way to reset all the settings to defaults without losing the essential settings in /cgi-bin/luci/admin/system/openmptcprouter ?

Because I'm working on this remotely, I can't just start again with a clean install.

Ysurac, 2022-08-31:
No way to reset and keep some settings.
For me the problem is only on this wan5. What do you have in OpenMPTCProuter status page about this gateway ? detected as working or as down ?

hairy-dog, 2022-08-31:
As a separate idea, I would find it really useful to have a way to simply set a few basic parameters in a text file: the details of the vps and the ip address settings of the router and the wan connections. Then have that file in the root of the sim card so when it boots, it picks those up. That would make installing much easier. I'm not sure how possible it would be to implement.

Back to the probem in hand:

The System / OpenMPTCProuter / status page shown all three wan connections as green, and the MPTCP support check brings back sender's key for all three, but now I am seeing VPN tunnel DOWN in amber in tbe router status section.

The Network / MPTCP / Bandwidth chart isn't showing the data that is flowing.

closed 2022-09-06 by hairy-dog