Ysurac/openmptcprouter #2501: WhatsApp Can't Make Call

ccmks, 2022-08-31:
Ever since I am using v2ray on version 5.85, and now using 5.9, still experience problem with calling in WhatsApp.

It will just simply connecting when attempt to answer the call.

If using other method is ok, but poor voice quality

What can be done to fix it?

Thank you

tuanbombt, 2022-08-31:
any function call with RTC had problem.

I try with Voice with discord, can't connect to Voice WebRTC.

Brazzo978, 2022-08-31:
Same problem when using v2ray alone some realtime voice protocol wont work

Ysurac, 2022-08-31:
Update to v0.59.1 and can you check using https://networktest.twilio.com/ that all is ok or not ?

ccmks, 2022-08-31:
Do I need to update the VPS when updating to 0.59.1?

ccmks, 2022-08-31:
Also, I couldn't find the 0.59.1 version download, need for Hyper-V Squashfs EFI image

Ysurac, 2022-08-31:
Yes you need to update VPS to 0.1028.
For the 0.59.1 you can use System->Sysupgrade menu or download the .img.gz file from https://www.openmptcprouter.com/download to update via System->Backup / Flash firmware menu.

ccmks, 2022-08-31:
I have pushed the update, now both OMR and VPS is on 0.59.1 and 0.1028 respectively. I will try to make WhatsApp call when I am on site tomorrow.

Thank you

Kalimeiro, 2022-08-31:
All is working great on 0.59.1, no problem with WhatsApp or Signal or Discord or ... and twilio test is OK.

So your problem is elsewhere.

obi12341, 2022-08-31:
i am on 0.59.1 and don't have problems
https://networktest.twilio.com/  <- everything is green

ccmks, 2022-09-01:
I have tested it after update to 0.59.1 and it is running good. Now I am using v2ray and I can make call to WhatsApp already. Unlike the previous version where it is just keep reconnecting

What's the difference between this 0.59.1 compare to previous version?

Kalimeiro, 2022-09-01:
> I have tested it after update to 0.59.1 and it is running good. Now I am using v2ray and I can make call to WhatsApp already. Unlike the previous version where it is just keep reconnecting
> 
> What's the difference between this 0.59.1 compare to previous version?

see changelog, lot of corrected bug and improvement, support of new hardware, debian 11 ready, 5.15 with mptcp nextgen...

Ysurac, 2022-09-01:
Short changelog is on https://www.openmptcprouter.com/

```
Changes:
* OMR-ByPass fixes
* SIP ALG fixes and it's enabled by default like release <0.59
* Various small fixes

VPS script need to be updated to 0.1028
```
Full changelog is https://github.com/Ysurac/openmptcprouter-feeds/compare/v0.59...v0.59.1

ccmks, 2022-09-21:
I just run into same issue now. WhatsApp keep reconnecting when receiving incoming call. I am on V2RAY

closed 2022-12-26 by github-actions[bot]