Ysurac/openmptcprouter #2507: no token yet available since 0.59.1 and 0.1028 update

vinceducat, 2022-09-02:
no token yet available since 0.59.1 and 0.1028 update
## Expected Behavior

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

Ysurac, 2022-09-03:
Check Status->System log to know why it can't get a token.

vinceducat, 2022-09-03:
sorry i reinstall from sractch the vps and its ok ...

closed 2022-09-03 by vinceducat

hairy-dog, 2022-10-31:
I am seeing the same problem. 

Mon Oct 31 23:31:26 2022 user.notice Shadowsocks: Rules DOWN
Mon Oct 31 23:31:28 2022 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 85.x.x.x, no server API answer on 85.x.x.x)

I have tried reinstalling the VPS but without wiping it first. It didn't help.

I think I shall have to try a clean start tomorrow by wiping the VPS and installing from scratch unless there any other things I could try.

Ysurac, 2022-11-01:
@hairy-dog you should read this issue: https://github.com/Ysurac/openmptcprouter/issues/2605 and if it's not the same open a new issue.

hairy-dog, 2022-11-03:
A clean install of the VPS fixed the problem. Seems like an upgrade had left something a bit wrong.

Fffrank, 2022-12-28:
I've had the same issue with an upgrade.  Starting from scratch with a clean VPS install also fixed it.