Ysurac/openmptcprouter #2509: NAT64 always on ? since 0.59.1 and 0.1028

vinceducat, 2022-09-03:
seem that NAT64 allways on even i desactivate in the wizzard menu
## Expected Behavior
try to disable NAT64
## Current Behavior
seems that nat64 couldnot be disable...

## Possible Solution
i dont know

## Steps to Reproduce the Problem

  1. check or uncheck quote square in the menu wizard 
  2. 
  

## Context (Environment)

milkywan vm vps provider

## Specifications

  - OpenMPTCProuter version: 059.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: milkywan french provider
  - OpenMPTCProuter platform: x86_64
  - Country: france

Ysurac, 2022-09-03:
What checkbox you mean ? If IPv6 is disabled then there is no IPv6 provided by OpenMPTCProuter. Can be provided by another box.

vinceducat, 2022-09-03:
in the OMR web wizard page under advance setting ipv6 section
"activer DNS64" option this is disable but seems to be always enable ...

Ysurac, 2022-09-03:
DNS64 only modify unbound configuration, and when it's disabled unbound DNS configuration.
What is the the result of "uci show unbound.ub_main" via SSH on the router when disabled ?

vinceducat, 2022-09-03:
redo update via loading last img seems to be ok now

vinceducat, 2022-09-03:
root@OpenMPTCProuter:~# uci show unbound.ub_main
unbound.ub_main=unbound
unbound.ub_main.add_extra_dns='0'
unbound.ub_main.add_local_fqdn='1'
unbound.ub_main.add_wan_fqdn='0'
unbound.ub_main.dhcp_link='none'
unbound.ub_main.dhcp4_slaac6='0'
unbound.ub_main.dns64_prefix='64:ff9b::/96'
unbound.ub_main.domain='lan'
unbound.ub_main.domain_type='static'
unbound.ub_main.edns_size='1280'
unbound.ub_main.extended_stats='0'
unbound.ub_main.hide_binddata='1'
unbound.ub_main.localservice='1'
unbound.ub_main.manual_conf='0'
unbound.ub_main.num_threads='1'
unbound.ub_main.query_minimize='0'
unbound.ub_main.query_min_strict='0'
unbound.ub_main.rebind_localhost='0'
unbound.ub_main.rebind_protection='1'
unbound.ub_main.resource='default'
unbound.ub_main.root_age='9'
unbound.ub_main.ttl_min='120'
unbound.ub_main.unbound_control='0'
unbound.ub_main.validator='0'
unbound.ub_main.validator_ntp='1'
unbound.ub_main.verbosity='1'
unbound.ub_main.trigger_interface='lan' 'wan'
unbound.ub_main.listen_port='5353'
unbound.ub_main.enabled='1'
unbound.ub_main.recursion='aggressive'
unbound.ub_main.protocol='mixed'
unbound.ub_main.dns64='0'

closed 2022-12-08 by github-actions[bot]