Ysurac/openmptcprouter #2510: Open ports using bypass omp

dgot86, 2022-09-04:
good morning everyone. i have this problem. a pc 192.168.1.31 must use ports 4662 tcp and 4682 utp and must pass only through a specific wan, in particular wan1 without going into VPN. I configured the bypass and so the pc comes out on the internet with the ip of the wan1 and not the vpn. works well. the problem is that the ports are not open. in the firewall section I configured the opening of the ports. on the wan1 modem I set the dmz to OpenMPTCProuter. I don't understand what's missing or where I'm wrong. if I use the VPN the ports are correctly open.
sorry my bad english.

vinceducat, 2022-09-04:
do you see the documentation example  to redirect a port.. ?
perhaps your provider block those ports

dgot86, 2022-09-04:
Yes.
With isp modem no problem to open ports.
With vpn no problem to open ports.
With bypass not work.

closed 2022-12-08 by github-actions[bot]