Ysurac/openmptcprouter #2511: Can't install new packages on Raspberry Pi 3B+, incompatible with the architectures issue

fsufyan, 2022-09-04:
## Expected Behavior
Installing new packages should succeed.

## Current Behavior
This message appears:
[image]

## Specifications
[image]

Ysurac, 2022-09-04:
I'm not able to reproduce the problem on a RPI4, can you check path in System->Software, "opkg configuration" ?

fsufyan, 2022-09-04:
This is the opkg configuration:
[image]

Ysurac, 2022-09-04:
You are using 64 bits image ?
libc is a base package, can you check the release you have in installed tab ?

fsufyan, 2022-09-04:
I downoaded the upgrade from this [link](https://download.openmptcprouter.com/release/v0.59.1/rpi3/targets/bcm27xx/bcm2710/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-bcm27xx-bcm2710-rpi-3-squashfs-sysupgrade.img.gz).

The other strange issue i have is that there is no installed packages shown on the opkg installed tab:
[image]

fsufyan, 2022-09-05:
This is the architecture when logging in router through terminal
[image]

fsufyan, 2022-09-05:
Flashed the pi without a sysupgrade image and now working, i guess something happened during the upgrade to v0.59.

closed 2022-09-05 by fsufyan