Ysurac/openmptcprouter #2512: Some data is missing when uploading data via FTP.

skmshd, 2022-09-04:
I created a file with 10000 lines of 0123456789 (file size: 110000byte) and uploaded it via passive FTP to check FTP behavior.
I don't know why but I observed missing data sometimes, and want to know how to fix it.

## Expected Behavior
Of course I expect to upload the file without any issues.

## Current Behavior
However, what I observed was the uploaded file had some missing data almost everytime. There is my test results:
- 1st: 66112byte
- 2nd: 109720byte
- 3rd: 110000byte
- 4th: 101712byte
- 5th: 107504byte

BTW, I used BYPASS option to bypass "fpt_control" and "ftp_data", but it didn't work. If I enabled this, I always see "Data connection timed out." error.

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0
  - OpenMPTCProuter VPS version: 0.1026.5.4.100-mptcp
  - OpenMPTCProuter VPS provider: vultr
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-09-04:
Update to latest release.
ByPass by service/protocol doesn't always work.

skmshd, 2022-09-06:
Thanks, sure, I will try the latest version and share result here.

skmshd, 2022-09-10:
Hi, I have updated to the latest version (v.0.59.1-54) but the same issue exists.. Anything else I can try to fix this?

Ysurac, 2022-09-10:
Did you try to bypass the FTP server IP ?

skmshd, 2022-09-11:
Hi Ysurac, thanks for your advice, IP based by-pass worked! Thanks, so I will close this, but one question.

The issue I reported (missing data), is this expected behavior ?

closed 2022-12-16 by github-actions[bot]