Ysurac/openmptcprouter #2513: How do I enable MPTCP over VPN

mbahmodin, 2022-09-05:
## Expected Behavior

Checkmark the MPTCP over VPN in, system > OpenMPTCPRouter, and it work

## Current Behavior

MPTCP over VPN: off
```
modem 1 (192.168.11.1)
Multipath seems to be blocked on the connection
ip address: 192.168.11.10
wan address: 182.2xxxxxx
interface: wan1
latency: 29 ms
mtu: 1500
multipath: master
```
Log: https://gist.github.com/mbahmodin/d69452eb4d44e26f251bf9313d60de60

MPTCP over VPN: on
MPTCP over VPN settings: WireGuard
It gives me an error
```
wgwan1 (10.255.247.1)
Gateway DOWN
No Server ping response after 1 second
ip address: 10.255.247.3
wan address: 182.2xxxxxx
```
and
```
wgwan1 (10.255.247.1)
No IP defined
```
Log: https://gist.github.com/mbahmodin/fbc54b68c5551ae8f572cc3d7d0d88d4

MPTCP over VPN: on
MPTCP over VPN settings: OpenVPN
```
No WAN with multipath enabled:

ovpnwan1
No IP defined
No gateway defined
```
Log: https://gist.github.com/mbahmodin/7733ef7929db05a22eb44dec17deae0d

## Specifications

  - OpenMPTCProuter version: 5.4.100-mptcp 0.1026 
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp 
  - OpenMPTCProuter VPS provider: IDCloudHost
  - OpenMPTCProuter platform: RPI4

Ysurac, 2022-09-05:
Update VPS to latest release (0.1028)
Without using MPTCP over VPN, connections are not aggregated ?

mbahmodin, 2022-09-05:
VPS server version: 5.4.100-mptcp 0.1028
router version: v0.59.1-5.4 r0+16594-ce92de8c8c

MPTCP over VPN: off
```
modem 1 (192.168.14.1)
Multipath seems to be blocked on the connection
ip address: 192.168.14.10
wan address: 182.2xxxxxx
interface: wan1
whois: TELKOMSEL-ID
latency: 45 ms
mtu: 1500
multipath: master
```
Log: https://gist.github.com/mbahmodin/285118f5462e35e72f5021603054a8d4

MPTCP over VPN: on Wireguard
```
wgwan1 (10.255.247.1)
No Server ping response after 1 second
ip address: 10.255.247.3
wan address: 182.2xxxxxx
mtu: 1420
multipath: master
```
Log: https://gist.github.com/mbahmodin/e42f00ec8ccdae9bdc72a65c5e6e4e7e

MPTCP over VPN: on OpenVPN
```
No WAN with multipath enabled:

ovpnwan1
No IP defined
No gateway defined
multipath: on
```
Log: https://gist.github.com/mbahmodin/ca076855484f8fbc8eaac0f270505887

mbahmodin, 2022-09-05:
MPTCP over VPN: off

omr-iperf
```
root@OpenMPTCProuter:~# omr-iperf vps -R 103.171.xxx.xxx
Connecting to host 103.171.xxx.xxx, port 65400
Reverse mode, remote host 103.171.xxx.xxx is sending
[  5] local 192.168.14.10 port 10063 connected to 103.171.xxx.xxx port 65400
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   140 KBytes  1.14 Mbits/sec                  
[  5]   1.00-2.00   sec   111 KBytes   910 Kbits/sec                  
[  5]   2.00-3.00   sec   168 KBytes  1.38 Mbits/sec                  
[  5]   3.00-4.00   sec   133 KBytes  1.09 Mbits/sec                  
[  5]   4.00-5.00   sec   140 KBytes  1.14 Mbits/sec                  
[  5]   5.00-6.00   sec   150 KBytes  1.23 Mbits/sec                  
[  5]   6.00-7.00   sec  74.6 KBytes   611 Kbits/sec                  
[  5]   7.00-8.00   sec  69.1 KBytes   566 Kbits/sec                  
[  5]   8.00-9.00   sec  52.9 KBytes   433 Kbits/sec                  
[  5]   9.00-10.00  sec   111 KBytes   911 Kbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.04  sec  1.64 MBytes  1.37 Mbits/sec   15             sender
[  5]   0.00-10.00  sec  1.12 MBytes   942 Kbits/sec                  receiver

iperf Done.
root@OpenMPTCProuter:~# 
```
omr-test-speed
https://gist.github.com/mbahmodin/c67bc8f2c3d9f4647e9660208510bffc

mbahmodin, 2022-09-05:
MPTCP support check
```
1: 103.171.84.172 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize -TCPOptionMPTCPCapable 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 5e:24:a2:e5:38:e7 , SourceMAC = 50:0f:f5:e4:33:d0 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 44 , Identification = 0xad44 , Flags = 2 , FragmentOffset = 0 , TTL = 251 , Protocol = 0x6 , CheckSum = 0x477d , SourceIP = 103.171.84.172 , DestinationIP = 192.168.14.10 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 44332 , SeqNumber = 229882945 , AckNumber = 1054629304 , DataOffset = 6 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 1608 , CheckSum = 0xf4fc , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1400 , >
```

MPTCP fullmesh
```
Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 4
3, 4, 0, 192.168.14.10, 571
IPv6, next v6-index: 0
```

MPTCP monitoring
https://gist.github.com/mbahmodin/01e26c7875f6aa1fe9032cd07b88fe31

Established connections
```
Error
```

Ysurac, 2022-09-05:
Set the interface you want to use with MPTCP over VPN as MPTCP enabled and select another interface to be MPTCP master.
To set correctly the VPN, the router part need to be able to contact VPS API, this can fail if the interface set as MPTCP over VPN is the master interface.

mbahmodin, 2022-09-05:
Like this?
I'm so confused right now
do I need to enter the static IP? because when I set it to "other" in the protocol option and after applying the setting, it goes back to static IP with an empty "IPv4 address" and "IPv4 gateway"
both interfaces show an error, but when I tested, I got internet access.
[image]
MPTCP Support Check: wgwan2
```
1: 10.255.247.1 IP::DiffServicesCP IP::CheckSum 
2: * 
3: 103.171.84.172 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 62 , Protocol = 0x6 , CheckSum = 0x7e62 , SourceIP = 103.171.84.172 , DestinationIP = 10.255.247.7 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 61828 , SeqNumber = 2475330199 , AckNumber = 812395082 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0x96fc , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 12050342471278459518 , >
```
[image]
MPTCP Support Check: wgwan2
```
1: 10.255.247.1 IP::DiffServicesCP IP::CheckSum 
2: * 
3: 103.171.84.172 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 62 , Protocol = 0x6 , CheckSum = 0x7e62 , SourceIP = 103.171.84.172 , DestinationIP = 10.255.247.7 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 19221 , SeqNumber = 2808222553 , AckNumber = 506893299 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0xb909 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 16528780936059969247 , >
```
[image]

mbahmodin, 2022-09-05:
I also tested a router with multiple wan in it.
before:
[image]
after adding a single interface with MPTCP over VPN with blank ip4 address and gateway
[image]
MPTCP Support Check wgwan7
```
1: 10.255.247.1 IP::DiffServicesCP IP::CheckSum 
2: * 
3: 103.171.84.172 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 62 , Protocol = 0x6 , CheckSum = 0x7e5e , SourceIP = 103.171.84.172 , DestinationIP = 10.255.247.11 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 29491 , SeqNumber = 3645551402 , AckNumber = 866327230 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0x25d6 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 15039349224327419376 , >
```

mbahmodin, 2022-09-06:
I'm testing adding another LTE modem with a different ISP that didn't block MPTCP. using this setting, but I don't know if I'm doing it right

my wizard setting
**wan1**
MacVlan
Physical interface: eth0
Multipath TCP: enabled
Enable SQM: on
Enable SQM autorate on
**wan2**
normal
Physical interface: wan2
DHCP
Multipath TCP: enabled
MPTCP over VPN: on
Enable SQM: on
Enable SQM autorate on
**wan3**
MacVlan
Physical interface: eth0
Multipath TCP: Master
Enable SQM: on
Enable SQM autorate on
All router settings: https://gist.github.com/mbahmodin/ce1f3b46338c8566d65879163dc21bbd
[image]
[image]

closed 2022-12-11 by github-actions[bot]