Ysurac/openmptcprouter #2516: OMR VPS debian 10 best way to update to debian 11 - advices

vinceducat, 2022-09-05:
Needs advices to update OMR VPS debian 10 to 11.
## Expected Behavior
update OMR VPS to debian 11 and have an operational VPS after this update ;-)
## Current Behavior
stay advice to do this update

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: Milkywan (FR)
  - OpenMPTCProuter platform: x86_64

Ysurac, 2022-09-06:
Only run the script.
But I think Milkywan still use Debian 10. The script doesn't update the OS (it was the case only for Debian 9 to Debian 10)

vinceducat, 2022-09-06:
@Ysurac can i do an debian 10 ti debian 11 update over OMR VPS install...

i thinks this wiil "broke" OMR install and after install relaunch omr scipt install

or it is better to install the VPS from scratch in debian 11 and re do OMR VPS install ?

Ysurac, 2022-09-06:
Update Debian 10 to Debian 11 should work, and yes launching install script can be needed to add some repo dependencies.

closed 2022-12-11 by github-actions[bot]