Ysurac/openmptcprouter #2518: Build version shows incorrectly "0.59-5.4"

ovdiyuk, 2022-09-06:
## Current Behavior
On UI there is the message"You have OpenMPTCProuter version 0.59-5.4 and version 0.59.1 is now available"

## Possible Solution
Correct versioning in the next build

## Steps to Reproduce the Problem

  1. Login in the UI
  2. On top of the screen under Heaser there is the message about a new version
  3. Go System -> Sysupgrade
  4. Upgrade the firmware
  5. Router will reboot many times and it will show again that new firmware is available.

## Context (Environment)
It looks like a mistake inside on version tag inside of the build

## Specifications
Version Installed: openmptcprouter-v0.59.1-5.4 from https://www.openmptcprouter.com/download
Model: Linksys WRT32X

obi12341, 2022-09-06:
duplicate of #2499

ogmkp, 2022-10-22:
If this is a duplicate of a previous ticket, can @ovdiyuk close this one please ?

ovdiyuk, 2022-11-02:
Sure. Thanks a lot for your work!

closed 2022-11-02 by ovdiyuk