Ysurac/openmptcprouter #2520: Is SSH needed for OpenMPTCP on the VPS side?

DigitalLabs420, 2022-09-07:
Hello everyone! 

Just a quick questions regarding the configuration and access of SSH from the VPS side. 

Does OpenMPTCP require this port to be bonded to 65222, or can I disable it completely without any side effects? 

On that note, would anyone have any tips to adjust shorewall to drop or reject SSH connections on the VPS. 

Thanks!

Ysurac, 2022-09-07:
You can disable it.
```DROP   net    $FW    tcp   65222``` (or all instead of net to block from vpn too)
```systemctl stop sshd``` to stop SSH server

You must have another way to manage the VPS.

DigitalLabs420, 2022-09-07:
Hi @Ysurac 

Really appreciate it! 

I do have other ways to manage the VPS.

closed 2022-09-08 by DigitalLabs420