Ysurac/openmptcprouter #2521: NTP timeout sync issues with time.windows.com

DigitalLabs420, 2022-09-07:
## Expected Behavior
Windows based system should be able to sync with time.windows.com or any other NTP time server 

## Current Behavior
Various Windows systems will timeout when trying to sync time with windows.time.com or any other NTP provided server. 

## Possible Solution
Maybe thre's something to do with NTP and the firewall, maybe it requires OMR bypass similar to how Disney.com gets affected? 

## Context (Environment)
Two OpenWRT Routers in Dual NAT --> Switch --> Desktop workstation  

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 
  - OpenMPTCProuter VPS provider: Binary Lane 
  - OpenMPTCProuter platform: x86_64
  - Country: Australia

DigitalLabs420, 2022-09-08:
I'll continue investigating when I have time, just wondering if people have experienced something similar

Ysurac, 2022-09-09:
I'm not using windows, but I can use this ntp server or any other.
Did you try on your VPS to use this ntp server ?

DigitalLabs420, 2022-09-12:
Hi @Ysurac, 

Thank you for your patience, been having a tough time trying to figure out why this isn't working. 

I've set the VPS's time server to windows.time.com and that appears to be working; however, any windows-based system using this server behind the router can’t sync. 

The time server issues may be the culprit in #2530. 

Since my setup is in Dual NAT, I temporarily disconnected the MPTCP router and wanted to see if I could sync with this time server using my regular Openwrt Router, which was successful. 

When I reconnected everything back into Dual NAT, the time server syncing issues re-occurred. 

I'm honestly at my wits end. I've tried manually adjusting DNS, adding exemptions into OMR-BYPASS, but no luck. 

I do appreciate your prompt response @Ysurac, I've attached images to outline the issues I've been experiencing. 

Thanks 

[image]
[image]
[image]

DigitalLabs420, 2022-09-14:
I'm just going to re-install and try non-dual NAT and see if the issue resolves. I'll provide an update on my current progress when I have some spare time.

closed 2022-12-19 by github-actions[bot]