Ysurac/openmptcprouter #2524: How to bind the interface to the 4G modem

sanfweb, 2022-09-08:
I use openwrt with 2 E3372 hilink modems, I disabled DHCP in the modems and assign a static IP for connection.

one modem in subnet 192.168.10.x the other in 192.168.20.x

if I connect modems in turn, first the first 192.168.10.x and eth1 is assigned to it, then the second 192.168.20.x is assigned to eth2, everything works fine, but if I insert them at the same time, maybe 192.168.10.x is assigned eth2 and vice versa, 192.168.20.x is assigned eth1 and then openwrt cannot connect to modems, the static address settings are not suitable, the modems have been swapped.

So, for example, it happens when electricity goes out and when you turn on the modems, the wrong eth is often assigned.

Is it possible to somehow hard-link which eth each of the devices receives, for example by IMEI or serial number of the device?

ubus call system board

`
 "kernel": "5.4.194",
        "hostname": "OpenMPTCProuter",
        "system": "ARMv8 Processor rev 3",
        "model": "Raspberry Pi 4 Model B Rev 1.1",
        "board_name": "raspberrypi,4-model-b",
        "release": {
                "distribution": "openmptcprouter",
                "version": "v0.59-5.4",
                "revision": "r0+16594-ce92de8c8c",
                "target": "bcm27xx/bcm2711",
                "description": "openmptcprouter v0.59-5.4 r0+16594-ce92de8c8c"`

uci export network

`config interface 'loopback'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'
        option multipath 'off'
        option device 'lo'
        option metric '5'

config globals 'globals'
        option multipath 'enable'
        option mptcp_path_manager 'fullmesh'
        option mptcp_scheduler 'blest'
        option congestion 'cubic'
        option mptcp_checksum '0'
        option mptcp_debug '0'
        option mptcp_syn_retries '2'
        option mptcp_subflows '3'
        option mptcp_add_addr_accepted '1'
        option mptcp_add_addr_timeout '120'
        option mptcp_fullmesh_num_subflows '1'
        option mptcp_fullmesh_create_on_err '1'
        option mptcp_ndiffports_num_subflows '1'
        option ula_prefix 'fd2b:d41d:d511::/48'

config interface 'lan'
        option proto 'static'
        option ipaddr '192.168.100.1'
        option netmask '255.255.255.0'
        option device 'eth0'
        option ifname 'eth0'
        option ipv6 '0'
        option delegate '0'
        option addlatency '0'
        option multipath 'off'
        option ip4table 'lan'
        option metric '6'
        option defaultroute '0'
        option peerdns '0'

config rule 'lan_rule'
        option lookup 'lan'
        option priority '100'

config interface 'wan1'
        option proto 'static'
        option ip4table 'wan'
        option multipath 'master'
        option defaultroute '0'
        option addlatency '0'
        option type 'macvlan'
        option masterintf 'eth0'
        option ipv6 '0'
        option metric '3'
        option peerdns '0'
        option ipaddr '192.168.10.100'
        option netmask '255.255.255.0'
        option gateway '192.168.10.1'
        option device 'eth1'

config device 'wan1_dev'
        option txqueuelen '20'
        option name 'wan1'
        option type 'macvlan'
        option ifname 'eth0'
        option mode 'vepa'
        option macaddr 'd0:7b:49:ab:f7:7c'

config interface 'wan2'
        option proto 'static'
        option ip4table 'wan'
        option multipath 'on'
        option defaultroute '0'
        option addlatency '0'
        option type 'macvlan'
        option masterintf 'eth0'
        option ipv6 '0'
        option metric '4'
        option peerdns '0'
        option ipaddr '192.168.20.100'
        option netmask '255.255.255.0'
        option gateway '192.168.20.1'
        option device 'eth2'

config device 'wan2_dev'
        option txqueuelen '20'
        option name 'wan2'
        option type 'macvlan'
        option ifname 'eth0'
        option mode 'vepa'
        option macaddr '30:01:a0:a7:2c:2b'

config interface 'omrvpn'
        option device 'tun0'
        option ip4table 'vpn'
        option multipath 'off'
        option leasetime '12h'
        option type 'tunnel'
        option txqueuelen '100'
        option metric '1200'
        option proto 'none'

config interface 'omr6in4'
        option proto '6in4'
        option ip4table 'vpn'
        option multipath 'off'
        option ipaddr '10.255.255.2'
        option peeraddr '10.255.255.1'
        option auto '0'
        option metric '1201'
        option ip6addr 'fe80::a00:2/126'
        option gateway 'fe80::a00:1/126'

config device 'lan_dev'
        option name 'eth0'`

Ysurac, 2022-09-09:
Check in System->OpenMPTCProuter, "Advanced settings" tab that "Disable interfaces auto rename" is unchecked.

sanfweb, 2022-09-09:
thanks a lot, it helped

closed 2022-12-14 by github-actions[bot]