Ysurac/openmptcprouter #2526: Unable to access the Luci web panel of the raspberry router

bluenor, 2022-09-08:
**Hello everyone Thank you so much for giving us the opportunity to take advantage of this fantastic project.**

For a while now, although it makes me connect to the internet, I can no longer access the web control panel of my raspberry router at 192.168.100.1 after login i can see only this

[image]

I have tried restarting the router but the problem persists. I don't know how to debug and I don't know how to check the logs to see what's wrong.

Ysurac, 2022-09-08:
Please complete the template when you open an issue.

Log are available via ssh using "logread".
Did you try another browser or a reboot ?

bluenor, 2022-09-11:
Yes you are right sorry.

1. Logs are available and i attach the file of log in this reply.
2. I tried another browser and nothing to do sadly.
[tmp.log](https://github.com/Ysurac/openmptcprouter/files/9543018/tmp.log)

thank you.

closed 2022-12-15 by github-actions[bot]