Ysurac/openmptcprouter #2527: Adding Weight to Interface

ccmks, 2022-09-09:
Hello,

I would like to suggest a feature that can be helpful for OMR. Adding weight to an interface that participate in MPTCP enabled. This is very useful when having connection that has data cap, but still want to participate in MPTCP. Here is the example

WAN1 - Fiber with 30 Mbps speed unlimited data cap
WAN2 - 4G LTE with daily data cap 5GB/day
WAN3 - 4G LTE with monthly data cap 30GB

If adding weight based on the interface, we can add ratio
WAN1 weight will be 7
WAN2 weight will be 2
WAN3 weight will be 1

That means the total packet will be 10 if you adding all weight on all WAN. The packet will be 70% flow on WAN1 and 20% flow on WAN2 and only 10% flow on WAN3

This feature might be useful on similar scenario too.

Thank you

Ysurac, 2022-09-09:
It was already suggested, but it's not so easy for now.
This will be made using "mptcpd" in the future.

ccmks, 2022-09-09:
OK, thank you for your consideration!

closed 2022-09-09 by ccmks