Ysurac/openmptcprouter #2528: Incorrect view of one box in the OMR Status(Luci)

Jacques-F, 2022-09-09:
## Expected Behavior

In the system / openmptcprouter / status view, all boxes give Name and local IP in the view.
I have 2 boxes (one ADSL and one 4G) and only the 4G box give a correct IP address.

## Current Behavior

In the system / openmptcprouter / status view, my Box has a false local IP.
I have 2 boxes (one ADSL and one 4G) and the first box give as IP address the 2 IP of DNS servers.
This 2 addresses are défined as static routes in the OMR configuration.

## Possible Solution

The name "Livebox (80.10.246.13081.253.149.1) is make with the 1st specific DNS server for Orange network, and whith the 2nd without space.
For sending mails with my postfix server, I must route the DNS query about SMTP to this DNS. If I use public DNS, I receive another IP address. For this, I declare spécific routes for this 2 servers (each server was routed to Livebox address).
## Steps to Reproduce the Problem

  1. Declare one box in the interface
  2. Manage one or more static routes through with address
  3. See the new name of the box in the page

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.58.5 
  - OpenMPTCProuter VPS version: 0.1027 
  - OpenMPTCProuter VPS provider: IONOS
  - OpenMPTCProuter platform: RPI4b 
  - Country: France 

[image]

Ysurac, 2022-09-10:
Can you update to latest release ? and if you have still the case I will need the result of "uci show network" and "ifstatus Livebox"

Jacques-F, 2022-09-10:
Hi,
You can close this bug... It's OK with the 0.59.1
Thanks for this program.

Jacques

closed 2022-09-10 by Jacques-F