Ysurac/openmptcprouter #2529: MacVLAN DHCP

himekifee, 2022-09-10:
**Is your feature request related to a problem? Please describe.**

When adding mptcp wan interface, only normal device has protocol DHCP support.

**Describe the solution you'd like**

Enable DHCP when choosing MacVLAN.

**Describe alternatives you've considered**

Manually add a macvlan device and use it when choosing base device. However, the existing device will be overwritten when config changed.

himekifee, 2022-09-10:
I'd also like a mac address setting for that as my internet provider requires mac registration. Thanks.

Ysurac, 2022-09-11:
DHCP on MacVLAN interface is not supported Why do you need this ?

himekifee, 2022-09-11:
By what means not supported? My apartment has ask4 restricting mac address. I'm trying to have single eth but multiple interfaces with different mac address to aggregation.

ccmks, 2022-09-13:
Why not using Static IP on each different WAN using single ETH? Forget about DHCP, it won't work on this setup

I am assuming you are trying to get DHCP as client for each WAN, right?

himekifee, 2022-09-13:
> 

No, it does not work with static IP. Their weird implementation forbids you. I tried but only have internet access through DHCP. I don't really wanna do reverse engineering on this. For your question, the answer is yes.

Ysurac, 2022-09-18:
You can set what you want using Network->Interfaces.

Ysurac, 2022-09-20:
At least you tried and confirm it's working ?

himekifee, 2022-09-20:
Well, not really. It showed all green but the link speed is only one portion. From ifconfig can see all traffic went through the first interface. Not sure if it is isp blocking. If I use mptcp over wg VPN, the mptcp check won't pass(without VPN won't pass either).

closed 2022-12-25 by github-actions[bot]